Infineon/XMC-for-Arduino #122: XMC4700 - DAC available?

ventilator, 2020-02-04:
Are the 2 DACs from the XMC4700 addressable from within Arduino? They are physically on the Pinheader X1
[image]

techpaul, 2020-02-04:
Basically YES, from the software, we need to find time to do an extended pinout diagram for XMC4700/XMC4800 boards. The DACs are assigned as
~~~
const uint8_t mapping_pin_DAC[][ 2 ] = {
                                        { 50, 0 },
                                        { 55, 1 },
                                        { 255, 255 } };
~~~
That is Arduino Pin number 50 and 55.

These are mapped to actual port numbers on chip as (/*nn */ is Arduino Pin number comment)
~~~
const XMC_PORT_PIN_t mapping_port_pin[] =
{
    /* 0  */    {XMC_GPIO_PORT2, 15},   // RX                           P2.15
...
	//Additional pins for port X1 starting here
	/* 36  */   {XMC_GPIO_PORT3 , 4},   // PWM42-2                      P3.4
	/* 37  */   {XMC_GPIO_PORT0 , 5},   // I2C_1 SDA  // SPI_4 MOSI     P0.5
	/* 38  */   {XMC_GPIO_PORT0 , 3},   // PWM80-2                      P0.3
	/* 39  */   {XMC_GPIO_PORT0 , 1},   // PWM80-1                      P0.1
	/* 40  */   {XMC_GPIO_PORT0 , 10},  //                              P0.10
	/* 41  */   {XMC_GPIO_PORT3 , 2},   //                              P3.2
	/* 42  */   {XMC_GPIO_PORT3 , 1},   //                              P3.1
	/* 43  */   {XMC_GPIO_PORT15 , 6},  // A6 / ADC Input               P15.6 (INPUT ONLY)
	/* 44  */   {XMC_GPIO_PORT15 , 5},  // A7 / ADC Input               P15.5 (INPUT ONLY)
	/* 45  */   {XMC_GPIO_PORT15 , 3},  // A8 / ADC Input               P15.3 (INPUT ONLY)
	/* 46  */   {XMC_GPIO_PORT14 , 15}, // A9 / ADC Input               P14.15 (INPUT ONLY)
	/* 47  */   {XMC_GPIO_PORT14 , 13}, // A10 / ADC Input              P14.13 (INPUT ONLY)
	/* 48  */   {XMC_GPIO_PORT14 , 7},  // A11 / ADC Input              P14.7 (INPUT ONLY)
	/* 49  */   {XMC_GPIO_PORT15 , 15}, // A12 / ADC Input              P15.15 (INPUT ONLY)
	/* 50  */   {XMC_GPIO_PORT14 , 9},  // DAC0 // A13 / ADC Input      P14.9 (INPUT ONLY)
	/* 51  */   {XMC_GPIO_PORT2 , 13},  //                              P2.13
	/* 52  */   {XMC_GPIO_PORT5 , 10},  //                              P5.10
	/* 53  */   {XMC_GPIO_PORT5 , 11},  // PWM80-0                      P5.11
	/* 54  */   {XMC_GPIO_PORT1 , 14},  //                              P1.14
	/* 55  */   {XMC_GPIO_PORT14 , 8},  // DAC1 // A14 / ADC Input      P14.8 (INPUT ONLY)
	/* 56  */   {XMC_GPIO_PORT15 , 14}, // A15 / ADC Input              P15.14 (INPUT ONLY)
~~~

So that is P14.8 and P14.9 for the DACs

Looking at board ident on reverse of PCB that is about 6th row down from the switches end of X1

Hope that helps

ventilator, 2020-02-04:
perfect, it works :-)
Here a simple demo DAC->ADC conversion

[image]

```
int analogPin = A0;

int analogOut = 50; // {XMC_GPIO_PORT14 , 9},  // DAC0 // A13 / ADC Input      P14.9 (INPUT ONLY)
int analogOut2 = 55; // {XMC_GPIO_PORT14 , 8},  // DAC1 // A14 / ADC Input      P14.8 (INPUT ONLY)

int val = 0;  // variable to store the value read

void setup() {
  Serial.begin(9600);           //  setup serial

  pinMode(analogOut, OUTPUT);  // sets the pin as output
  pinMode(analogOut2, OUTPUT);  // sets the pin as output
}

void loop() {
  for (int i=0; i<1023; i++) {
  
    analogWrite(analogOut, i);
    analogWrite(analogOut2, i);
    
    val = analogRead(analogPin);  // read the input pin
    Serial.println(val);          // debug value
  }
}
```

closed 2020-02-04 by ventilator

techpaul, 2020-02-04:
Please note that resolution of the ADC and DACs can be set to be 12 bit see 

analogReadResolution

analogWriteResolution

Pass in 8, 10 or 12 to change  the resolution size

See also the Wiki

https://github.com/Infineon/XMC-for-Arduino/wiki/Analog-Functions-and-Additions