Infineon/XMC-for-Arduino #124: USB/Mouse libraries are broken

danielstuart14, 2020-02-06:
Currently, building any of the example files for the USB or Mouse library results in a bunch of errors.
My guess is that the USBLib path isn't being set correctly. I have no expertise on Arduino's build system, which make it quite difficult for me to track the issue down. 

Some of the errors:
```
In file included from /Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/Descriptors.c:38:0:
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/Descriptors.h:76:4: error: unknown type name 'USB_Descriptor_Configuration_Header_t'
    USB_Descriptor_Configuration_Header_t    Config;
    ^
In file included from /Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/VirtualSerial.h:43:0,
                 from /Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/VirtualSerial.c:37:
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/Descriptors.h:76:4: error: unknown type name 'USB_Descriptor_Configuration_Header_t'
    USB_Descriptor_Configuration_Header_t    Config;
    ^
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/Descriptors.h:79:4: error: unknown type name 'USB_Descriptor_Interface_t'
    USB_Descriptor_Interface_t               CDC_CCI_Interface;
```

techpaul, 2020-02-06:
Important questions 

1.  Which Arduino  IDE version
2.  Which board are you compiling for as only XMC4xxx series support the USB functions
3.  Is this all examples or a specific example

techpaul, 2020-02-06:
If you are using the examples from "Built-in libraries" section of the examples menu under
       09. USB
The examples clearly state at the top of the files they are for
- Leonardo
- Due
- Micro

These are Arduino's own boards NOT compatible with ANY XMC board

danielstuart14, 2020-02-06:
Arduino IDE 1.8.10 on linux. I'm using a xmc4700 relax kit with usb support.
All the xmc USB/Mouse examples are throwing errors when compilling.

How to reproduce:
Select xmc4700 relax kit -> Examples -> Examples for xmc4700 relax kit -> USB/Mouse -> select any project and compile

danielstuart14, 2020-02-07:
Well, I solved this issue. But not in an elegant way.
What happens is that USB0 wasn't defined when USB library was including usblib.h.
This was easily fixed by including Arduino.h into the Descriptors.h file. Here: https://github.com/Infineon/XMC-for-Arduino/blob/0fa47ed19a65ace89f021b51b33c64ef414607cb/arm/libraries/USB/src/util/Descriptors.h#L42

But, although most of the errors were gone, there were still some errors while linking the files:
```
/var/folders/kf/5vlzlwnn6y38klgkvnlklnw80000gn/T/arduino_build_805491/libraries/USB/USB.cpp.o: In function `USBClass::begin(unsigned long)':
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/USB.cpp:38: undefined reference to `USB_DeviceState'
/var/folders/kf/5vlzlwnn6y38klgkvnlklnw80000gn/T/arduino_build_805491/libraries/USB/USB.cpp.o: In function `USBClass::IrqHandler()':
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/USB.cpp:132: undefined reference to `CDC_Device_USBTask'
Foram encontradas múltiplas bibliotecas para "USB.h"
Usado: /Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/USB.cpp:134: undefined reference to `CDC_Device_BytesReceived'
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/USB.cpp:136: undefined reference to `CDC_Device_ReceiveByte'
/var/folders/kf/5vlzlwnn6y38klgkvnlklnw80000gn/T/arduino_build_805491/libraries/USB/util/VirtualSerial.c.o: In function `USB_Init':
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/VirtualSerial.c:105: undefined reference to `USBD_Initialize'
/Users/danielstuart/Library/Arduino15/packages/Infineon/hardware/arm/1.4.0/libraries/USB/src/util/VirtualSerial.c:118: undefined reference to `device'
/var/folders/kf/5vlzlwnn6y38klgkvnlklnw80000gn/T/arduino_build_805491/libraries/USB/util/VirtualSerial.c.o:(.data.USB_runtime+0x4): undefined reference to `USBD_SignalDeviceEventHandler'
/var/folders/kf/5vlzlwnn6y38klgkvnlklnw80000gn/T/arduino_build_805491/libraries/USB/util/VirtualSerial.c.o:(.data.USB_runtime+0x8): undefined reference to `USBD_SignalEndpointEvent_Handler'
collect2: error: ld returned 1 exit status
```

Theses errors were "solved" by adding the gcc flag -DUSB0 to the boards.txt file. Here, to be exact: https://github.com/Infineon/XMC-for-Arduino/blob/aaee2f30d6e2a9755891f8912a9e5c28555dde07/arm/boards.txt#L202

This made me able to compile the USB examples, but also gave some warnings during compile time.
So, even though it works, I would like a way of accomplishing that without additional warnings.

danielstuart14, 2020-02-07:
Edit: Just saw that -DUSB0 was removed only on the development branch....

techpaul, 2020-02-07:
I personally have not had time to look at this (many other things on)

-DUSB0 compile flag is removed as it was later redefined as typedef in XMC4700.h, this causes the compiler confusion on which to use
~~~
#define USB0                            ((USB0_GLOBAL_TypeDef                *) USB0_BASE)
~~~
Probably need to look at history of XMC4700.h first.

The other issues seem related to how some of the circular includes are done and what the compiler is doing. This is similar to your issue on Serial1, which whilst I have a soultion that stops circular includes makes it easier to maintain for boards that later may have 4 serial ports.

In this case I see all sorts of thinsg suggesting Arduino.h is not being included correctly as I have seen errors like not knowing what a uint16_t is. The descriptors should be included from USB.h, some nesting of include files is causing confusion and wrong things being excluded. Some of this may be simple .ino file not being compiled as C++ but C, it could be other issues.

I think this is going to take a while to wrestle with the hierachy of include files and build options.
It may be worth checking against an older version that used the older compiler.

closed 2021-07-20 by techpaul