Infineon/XMC-for-Arduino #127: develop branch header files and Fix additional serial ports

techpaul, 2020-02-09:
Latest merge in develop branch rationalises core header includes reduces circular references and seems to improve situation

## Fix Serial1 (or any additional serial ports) external definition
This is related to a previous PR to tidy core modules header files and ordering. The issue was that
- NUM_SERIAL **IS defined** when compiling HardwareSerial **.cpp** (map file shows the classes, interrupt functions etc)
- NUM_SERIAL is **NOT defined** when including HardwareSerial **.h** as pins_arduino.h not included yet so missing
~~~
extern   HardwareSerial Serial1;
~~~
Simple fix is to move the part of HardwareSerial.h that makes external class declarations to pins_arduino.h .

Better fix implemented here is to move the board specific parts of HardwareSerial.h to pins_arduino.h, thus making
HardwareSerial a more generic class module that does not care how many ports there are. The number of ports are then
fully defined in pins_arduino.h, making boards that support different number of serial ports easier to create and **maintain** .

Making it easier to create 2 serial port XMC2GO or XMC1100 Boot Kit or even 4 serial port XMC4700 to easier implement

A 4 serial port XMC4700 Relax or even Relax Lite board makes a good alternative to Arduino Mega, Due or other multiple 
serial port boards (something I use often).

### Example to test with
An additional Example tree for multiple serial is also included that moves data from one port to another, so if
the final port (Serial1) has a loop back connection (TX to RX) simple testing can be done using one Serial Monitor
or terminal emulation window. This has been tested on an XMC4700 Lite (or for Arduino Shield 5V) with no ethernet or SD card fitted.

Example is called **TwoPort** under **Multi-Serial** allowing for later additions of Three or Four port examples.

techpaul, 2020-02-10:
See PR #126 and note develop branch for non-USB gives cleaner compile for warnings

techpaul, 2020-05-15:
This issue (announcement) can be closed on new version release

AlucardMode0, 2020-05-15:
ok . i can use Serial1.print

but it wont do anything on my pins 2.14 , 2.15 

only on Serial , pins for   
/* 34  */   {XMC_GPIO_PORT1 , 4},   // USB Debug RX                 P1.4
/* 35  */   {XMC_GPIO_PORT1 , 5},   // USB Debug TX                 P1.5

the debug pins

techpaul, 2020-05-15:
USB TX and RX are connected to debugger to go to host, to do host and 2 others needs THREE UARTs on Arduino framework

On Arduino (any board type) the default is first serial (serial) is to host.

You could look at a configuration chnage to remove an SPI or I2C and add a third Serial (Serial2)

techpaul, 2020-06-29:
Issue fix is in release V1.5.0 so closing

closed 2020-06-29 by techpaul