Infineon/XMC-for-Arduino #134: XMC2go can't find I2C devices

Creepygangster, 2020-03-31:
Hi,
I'm trying to control a lcd-display with a XMC2go, but it doesn't display anything.
I think the adress I'm using is wrong, but the I2C-scanner (https://playground.arduino.cc/Main/I2cScanner/) can't find any I2C-devices.
This is the small test code I'm trying to run.
#include <LiquidCrystal_I2C.h>
#include <Wire.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);

void setup()
{
  lcd.init();
  lcd.backlight();
}

void loop()
{
  lcd.setCursor(0,0);
  lcd.print("hello world");
}
  
Thanks in advance

techpaul, 2020-03-31:
There are MANY MANY possibilities
1/ We have no idea what the display is being written to via I2C
2/ XMC2Go is a 3V3 device and the display may be 5V only
3/ Lack of level shifter LIKE one as decribed on a link from that page https://playground.arduino.cc/Main/I2CBi-directionalLevelShifter/
4/ The library LiquidCrystal_I2C may be incomaptible as in meant for AVR processors only
5/ You could have a wiring fault (signals and/or power)
6/ Lack of correct pull-up resistors
7/ Other problems

I tend to think your most likely causes are (6), (5), (2) and (3)

Item (7) is always possible.

Creepygangster, 2020-03-31:
First Problem was: no pull-up resistors. I had this display running without them on an arduino before, so I didn't really think about them. Now I have 10k resistors as pull-ups.
Second Problem is from possibility 7: the contrast of the display was set too low. Now it works almost perfectly, the 3,3V from the XMC2Go are a little bit dim, but later I will have a 5V supply for the display, this is just a temporary setup for debugging.
Thanks for the quick answer.

closed 2020-05-15 by techpaul