Infineon/XMC-for-Arduino #139: Serial Output Selection is not working

djaumann, 2020-06-02:
The selection of the serial output is not working in the Arduino IDE. No matter what port you chose "PC" or "On Board" it seems like the build flag is not built. If you need more details let me know!

techpaul, 2020-06-02:
Yes we do and what simple application are you using to prove this?

See pinned article in issues on reporting issues #100 

This serious a problem MIGHT exist but if it is in current release it would have been reported many times before.

djaumann, 2020-06-02:
Right now I'm using the TLE5012b-Kit with the corresponding Git-Lib. I'm reading out the angle value and want to print that. But somehow I cannot get the values on the TX, RX pins with the Arduino IDE. I also tried this with PlatformIO with a build flag and that is working.

techpaul, 2020-06-02:
Please note the XMC-for-Arduino supports BASE XMC2Go see

https://github.com/Infineon/XMC-for-Arduino/wiki/XMC-2Go

The variant or any examples or libraries are not to my knowledge in the current support

mhollfelder, 2020-06-02:
Problem is related to how the flag is set in Arduino and how the pins_arduino.h checks it.
As shown [here](https://github.com/Infineon/XMC-for-Arduino/blob/0fa47ed19a65ace89f021b51b33c64ef414607cb/arm/boards.txt#L67), the flags are defined in the boards.txt of the master branch:

```
XMC1100_XMC2GO.menu.UART.debug=PC
XMC1100_XMC2GO.menu.UART.debug.flag.selected=SERIAL_HOSTPC
XMC1100_XMC2GO.menu.UART.onBoard=On Board
XMC1100_XMC2GO.menu.UART.onBoard.flag.selected=SERIAL_ONBOARD
```

In the develop branch, this has changed due to fixing platform.txt as well.
More details can be found in the current branch [here](https://github.com/Infineon/XMC-for-Arduino/blob/61709376bdd07003524d265d3f59537ab28067e9/arm/boards.txt#L35).

However, the flags which are checked in XMC2Go pins_arduino.h is shown [here] in th(https://github.com/Infineon/XMC-for-Arduino/blob/0fa47ed19a65ace89f021b51b33c64ef414607cb/arm/variants/XMC1100/config/XMC1100_XMC2GO/pins_arduino.h#L157) from the master branch:

```C++
#ifdef SERIAL_DEBUG
                            .pin  = (uint8_t)2
#else
                            .pin  = (uint8_t)6
#endif
[...]
#ifdef SERIAL_DEBUG
                            .pin  = (uint8_t)1
#else
                            .pin  = (uint8_t)0
#endif
```

The defines and board.txt are completely messed up, also for flipping the UART pins of XMC1100 Boot Kit. I will check later and fix it in the develop branch.

Best regards,

Manuel

djaumann, 2020-06-02:
Ah perfect, maybe I should have looked in the development branch. Ok then I hope you can fix it somehow. Thanks @mhollfelder for the quick response and also @techpaul for the help.

Best regards,

Dominik

techpaul, 2020-06-29:
develop branch fix has been incorporated into master and release V1.5.0 closing issue

closed 2020-06-29 by techpaul