Infineon/XMC-for-Arduino #141: set task for xmc4700

AlucardMode0, 2020-06-11:
hello everyone . i reading the wiring_time and tone files and i dont understand too much how can i make an task to call himself at ... idk  50hz or 100hz . Can anyone explain to me how to use them to create 2 task with same priority ?

techpaul, 2020-06-11:
For an example program find your Arduino15 folder then find folder

packages\Infineon\hardware\arm\<version>\tests

Alternatively open an existing example for Infineon, then select open, go up one folder to find tests folder.

Look for tonetest.ino a test project that checked opertions that will flash as many onboard LEDs that exist at 1Hz, 2Hz...... for varying durations, but first LED (LED1) will flash continuoulsy at 1 Hz, the only lines you need worry about are tone(...)

The serial port output is for testing the running tasks software the rest enables tones to run, mainly from hardware interrupts, using SYSTICK ms interrupt..

All tone pin tasks are checked EVERY millisecond to see if they need to be run all are at same priority (bar order of execution which is minor). When a match interval is found the necessary tone pin toggling is actioned,

Function  noTone(...) will stop any tone pin

techpaul, 2020-06-11:
If you are wanting to add your own timed taks that is a different matter and requires other functions and knowledge of the scheduling used along with handling of available free tasks in list with error handling.

closed 2020-06-11 by AlucardMode0