Infineon/XMC-for-Arduino #142: tasks of Tone

AlucardMode0, 2020-06-11:
from how much i understand the Tone is for generating a toggle on  pin at some freq ..
when i call in my Main the function tone(pin , freq, duration)(**_### it should start the task 0_**)

is setting the the tasks and tone_pins  ok , but 
[image]
running is a global variable , and its 0 ... 
and 

[image]
the running remains 0 

[image]

and when task 0 ( from tone) comes for start task  it imediatly -1 
[image]

not a valid running task . why is that ?

techpaul, 2020-06-11:
Where is yoiur TEST snippet of code (as text)?
Keep code snippets small

1. Have you tried tonetest.ino as directed above ?
2.  Once a tone is setup (usually in setup( )it runs continuously, continually calling in loop( ) means you are continuously run by HARDWARE and interrupt routines.
3. Screenshots of the LIBRARY code does not help anyone we can all pull up the files easier to read on our own screens, in readable formating.
4. Post code snippets as text using the '<>' button above

Please state EXACTLY what you are trying to achieve as a set of steps, then we can help you.

techpaul, 2020-06-11:
If you are trying to monitor global variables for task lists please note these variables are VOLATILE as they are updated in interrupt routines, which the debugging method and IDE you are using may not cope with correctly.

techpaul, 2020-06-11:
This is a continuation of Issue #141 please respond in the SAME issue do not create a new issue each time

AlucardMode0, 2020-06-11:
yes i have tried and the first call of tone function doesnt work no matter what :D . 
when the code starts , running is 0 , even if no task is running and when the task 0 is comming from tone, the checkid 
int CheckID( int ID )
{
if( ID < 0 || ID >= _MAX_TASKS )
  return -1;
if( ID == running )
  return 0;
return 1;
}

returns 0 even if a task is not running  

this error is occurring only at the first call of tone function , and from now i will close the other issue #141

techpaul, 2020-06-11:
WHAT IS YOUR CODE SETTING THIS UP

The library code for tones, milli-second counter and other functions has been working in that manner for over a year, so what is your code calling tone doing

techpaul, 2020-06-11:
See https://github.com/Infineon/XMC-for-Arduino/tree/master/arm/tests/tonetest

For tonetest.ino working on many boards.

PLEASE try this test code first and see the LEDs flashing to see this works

AlucardMode0, 2020-06-12:
https://photos.google.com/share/AF1QipPXjLg9eEgVaBz4MSjCpUV6EjmNmqkT28CNCBWag6Q9DXlbHBhOuUKNfdeEYDNP0g?key=TF91TG9aLXhnTU9pczJxTWJmX09VTGZZMHdyZDF3

1 st test was like this 

tone( LED1, 1 );            //this didnt work 
tone( LED2, 2, 20000 );// this one was blinking

and second like this 

tone( LED2, 1 , 5000 );// this didnt work
tone( LED1, 2, 20000 );//this one was blinking 

and everythig is from develop branch .
Conclusion . first call of tone function wont work cause is considered running before starting the execution . or from **running** variable from **wiring_time.c** or from function :
**static int8_t toneBegin( uint8_t _pin )** from **Tone.cpp** that starts from 0 to count the tasks.

when the task0 starts is (the tone task ) this occurs: 

int CheckID( int ID ) // id from first call of tone is 0
{
if( ID < 0 || ID >= _MAX_TASKS )
  return -1;
**if( ID == running )// running is 0 soo 
  return 0;// this returns 0** 
return 1;
}

int **startTask( int ID )**
{
// Check valid ID, not running and interval
**if( ( CheckID( ID ) ) <= 0 ) // ( CheckID( ID ) =0 ( the return) <= 0 soo 
  return -1;** //returns -1 "the task is running" . when it is not . 
if( taskTable[ ID ].interval == 0 )
  return -2;
if( taskTable[ ID ].status  > 0 )
  return -3;
// Start task
taskTable[ ID ].next = old_ms + taskTable[ ID ].interval;
taskTable[ ID ].status = 1;
return 1;
}

i made the test with the arduino IDE instead of Visual Studio Code  but there should be no diference .

techpaul, 2020-06-12:
Do NOT post your test code as a PHOTO I CANNOT read it When sending a comment copy and paste the code (and any other code into the comment box, highlight it and click the <> button above the comment box This will format the code as a code block that is readable

Alternatively make start of line be ~~~
paste the code in and then add a new line with just ~~~

Use preview tab to view it

Example result (which is readable)
~~~
int CheckID( int ID )
{
if( ID < 0 || ID >= _MAX_TASKS )
  return -1;
if( ID == running )
  return 0;
return 1;
}
~~~
As you are using your own code and not requested tonetest.ino to find what the issue is or is different circumstances. There is a POSSIBILTY of an EXTREME EDGE case, which I cannot confirm on the data you have supplied. Please use your existing test code but add the following at beginning of setup()
~~~
void setup()
{
delay( 20 );
...... existing setup coide
~~~

This may or may not confirm the EXTREME EDGE case

techpaul, 2020-06-15:
Having looked at possibilities I have a minor fix that will shortly be a PR into develop branch.

Tested using tonetest.ino on XMC4700 Relax LIGHT and run on board

The issue is a FAST CPU and requesting a tone **BEFORE** the **FIRST** Systick interupt has occured 1 ms after Power up/reset. This is an EDGE case specific to order of user code and Systick interrupt timing.

techpaul, 2020-06-15:
See Pull Request #143 as fix for this edge case

Will leave a week for comments before merging

techpaul, 2020-06-29:
As this has been incorporated and is now in V1.5.0 closing issue as fixed

closed 2020-06-29 by techpaul