Infineon/XMC-for-Arduino #149: XMC1400 Boot Kit definitions

techpaul, 2020-06-29:
pins_arduino.h for XMC1400 Boot Kit has at least two errors
~~~
#define XMC_BOARD           XMC1400 Boot Kit
~~~

Should be (as per other boards)
~~~
// XMC_BOARD for stringifying into serial or other text outputs/logs
// Note the actual name XMC and number MUST have a character between 
// to avoid issues with other defined macros e.g. XMC1100
#define XMC_BOARD           XMC 1400 Boot Kit
~~~
The build from boards.txt etc will set a compilation variable of _-DXMC1400_ this will cause problems on any use of stringifying as without the space between XMC and 1400 a substitution is made

Second issue is there is a LED library to simplify examples that need an LED to toggle, or go on or off. Unfortunately many boards have **DIFFERENT** methods for lighting an LED some use 0 for ON some use 1 for ON. To that end each pins_arduino.h for ALL other boards has one of the following lines in

XMC1100 Boot Kit
XMC1300 Boot Kit
~~~
/* On board LED is ON when digital output is 0, LOW, False, OFF */
#define  XMC_LED_ON         0
~~~
or

XMC1100 XMC2Go
~~~
/* On board LED is ON when digital output is 1, HIGH, TRUE, ON */
#define  XMC_LED_ON         1
~~~

Which one does XMC1400 Boot Kit supposed to have as I have no access as yet to circuit diagram for XMC1400 Boot Kit

jaenrig-ifx, 2020-06-30:
The "arduino_pins.h" of the xmc1400 is not matching the [schematic ](https://www.infineon.com/dgdl/Infineon-XMC1400_ArduinoKit-UserManual-v02_00-EN.pdf?fileId=5546d4626cb27db2016d48e4c7051906). The variants files must be just a placeholder.

techpaul, 2020-06-30:
@jaenrig-ifx  @mhollfelder XMC1400 Boot Kit addition looks a flop see why below 

> The "arduino_pins.h" of the xmc1400 is not matching the [schematic ](https://www.infineon.com/dgdl/Infineon-XMC1400_ArduinoKit-UserManual-v02_00-EN.pdf?fileId=5546d4626cb27db2016d48e4c7051906). The variants files must be just a placeholder.

Have you LOOKED at that *manual* which is not a schematic. In particular the pictures of the board ?
That manual is an XMC1100 Boot Kit Manual with search and replace of XMC1100 to XMC1400 throughout

Even your own site shows a completely differerent board shape and size for the XMC1400 Boot Kit see https://www.infineon.com/cms/en/product/evaluation-boards/kit_xmc14_boot_001/ I suspect the circuit is actually an XMC1100 Boot Kit. Interestingly the Product status shows **discontinued**

"arduino_pins.h" and the variants tree **CANNOT** be a placeholder otherwise NOTHING works, a bit like not having a main() You will not get any communications between board and host for Serial.print

What actual **PHYSICAL** tests have been done using XMC-for-Arduino and XMC1400 Boot Kit ???

If nothing else XMC-for-Arduino tests on an ACTUAL board should consist of running

tests\testdefines to see sensible figures and model numbers and serial debug comms
tests\tonetest to see scheduling and flashing LEDs (definitions of GPIO and LEDn correct

Neither require complicated equipment

techpaul, 2020-06-30:
A distributor of the boards has a manual that looks **MORE** like the actual board

http://www.farnell.com/datasheets/2631541.pdf?_ga=2.110334469.1093138544.1593506156-1760954961.1472845549&_gac=1.11734656.1593507231.EAIaIQobChMI2fSKhpGp6gIVGLLtCh2WFA3UEAAYASAAEgJCCPD_BwE

jaenrig-ifx, 2020-06-30:
Okay. My bad... 🤦 The KIT_XMC14_BOOT_001 is a different baord than the KIT_XMC1400_ARDUINO.

The board is still not enabled in the boards.txt. That should happen after some PHYSICAL verification ;) But exactly, it is discontinued... Anyhow the changes you suggested are already added to the "arduino_pins.h" on this pull request #152 .

techpaul, 2020-06-30:
Well lots of 4400, 1400 and variations  flyiing about.

Quick examine of the Arduino Kit manual shows it is a bad rewrite of XMC1100 Boot Kit manual (some sections actually say XMC1100 and things like 6 LEDs. Worst of all not handling Reset pin correctly and a fix was done for XMC1100 Boot Kit to remedy that and will need the same mods. Needs better resolution images for board and schematic as nearly unreadable.

In order to be verified there are many patches that can be done in readiness. I will start another issue on readiness for XMC1400 Arduino Kit. Also patches to change the boards.txt and other parts as best as possible to match Arduino Kit board
Many XMC1100 Boot Kit mods pertain to this board

Note manual says XMC1404-TO38 which is not listed in latest datasheet for device.

jaenrig-ifx, 2020-07-01:
I see now... Thanks for the feedback regarding the documentation. It has been forwarded to the responsible team :)

techpaul, 2020-07-01:
\I will look at doing a brand in my fork for XMC1400_Arduino_Kit as rename of XMC1400_Boot_Kit variant, put all known fixes that need to be in there from XMC1100 and other XMC1 boards with correct pin out to match the manual as best as possible.

This will give a starting point for any physical testing.

What is needed for XMC4400 2 GO is a Wiki page with pinout diagram

techpaul, 2020-07-01:
As main issues sorted now need to finish a few changes and will as soon as possible get a new test branch merged here for XMC1400 Arduino Kit

Closing this issue

closed 2020-07-01 by techpaul

jaenrig-ifx, 2020-07-02:
Sure. Only the wiki for the Platform 2Go is already there with the pinout diagram -> https://github.com/Infineon/XMC-for-Arduino/wiki/XMC4400-Platform2Go

techpaul, 2020-07-04:
Was not there at time I checked is now

I will see if XMC1400 pinout can be created in preparation, other issues to sort first