Infineon/XMC-for-Arduino #150: How can I use a  3rd party library expecting the 'usingInterrupt' function in class 'SPIClass'

lheusser, 2020-06-29:
Hi all

Here the system I am using:

System Part | Type / Version
------------ | -------------
Used Board | XMC 4700 Relax Kit Serie-V1
XMC-for-Arduino | 1.5.0
Arduino IDE | 1.8.12
Specific HW | MCP2518FD click (mikro BUS)
Specifics Libs | ACAN2517FD  / Ver. 2.1.2

I try to uses the ACAN2517FD library together with the XMC 4700 Relax Kit

```c
#include <ACAN2517FD.h>
void setup() {
}
void loop() {
}
```
but I have following compiler error:

```
...Arduino\libraries\ACAN2517FD\src\ACAN2517FD.cpp: In member function 'uint32_t ACAN2517FD::begin(const ACAN2517FDSettings&, void (*)(), const ACAN2517FDFilters&)':
...\Arduino\libraries\ACAN2517FD\src\ACAN2517FD.cpp:511:14: error: 'class SPIClass' has no member named 'usingInterrupt'
         mSPI.usingInterrupt (itPin) ; // usingInterrupt is not implemented in Arduino ESP32

```

For me it is not clear what the difference is in the SPI implementation on the XMC compared to the standard Arduino. Could you please give me some hint, how can I get the ACAN2517FD working with the XMC-Arduino?

Thanks in advance!

jaenrig-ifx, 2020-06-29:
Hi @lheusser,

It seems in the XMC-for-Arduino SPI API implementation, there is no such "usingInterrupt" function. You can try instead to the "attachInterrupt()" function.

https://github.com/Infineon/XMC-for-Arduino/blob/master/libraries/SPI/src/SPI.h
```
    // SPI Configuration methods
    void attachInterrupt();
    void detachInterrupt();
```

We keep the issue as an enhancement for interoperability with Arduino.

techpaul, 2020-06-29:
My first worry is that even though we can see that usinginterrupts not supported your error message shows compilation for a  **DIFFERENT** board namely an ESP32 board
~~~
   mSPI.usingInterrupt (itPin) ; // usingInterrupt is not implemented in Arduino ESP32
~~~
Whilst the original specification for Arduino has usinginterrupts() it would need some digging into how that worked even for HW-SPIor SW-SPI that is supported, Even the Arduino reference is a bit sparse in how to use it

https://www.arduino.cc/en/Reference/SPIusingInterrupt

Whilst ttachInterrupt() and detachInterrupt are listed in header file shown looking at https://github.com/Infineon/XMC-for-Arduino/blob/master/libraries/SPI/src/HW_SPI.cpp

The code is at least not "enabled"
~~~
void SPIClass::attachInterrupt()
{
    // Should be enableInterrupt()
}

void SPIClass::detachInterrupt()
{
    // Should be disableInterrupt()
}
~~~

Just noticed scanning files that the examples\BarometricPressureSensor also conatins a COPY of the example in
examples\BarometricPressureSensor\BarometricPressureSensor will sort out a pull request to tidy that up.

jaenrig-ifx, 2020-06-29:
Ok, I didn´t check further... 🤦. usingInterrupt() is used in the AVR Arduino implementation to avoid hardware interrupt conflicts with the SPI transactions. 
It would not make much sense to add such "usingInterrupt()" function for the xmc based SPI implementation. Only for API compatibility with AVR Arduino core based libs, keeping its definition empty. 

@lheusser I guess removing/commenting that line on the ACAN2517FD library could just do the work and have no impact on the SPI operation... You can give it a try.

True, that Barometric copy example needs to be removed.

techpaul, 2020-06-29:
Considering SPI controller is a full controller and has FIFO, the usinginterrupts is hangover of the *awkward* AVR architecture and should be made into null operations.

I would suggest
- Changing attach/detachinterrupt to usinginterrupts pair for compatability
- Checking what if anything uses SW-SPI as this is not standard and makes support more complicated, if possible drop it as well.

The SPI is a bit messy for compatability and following through.

Working through repo syncs and PR for duplicate and external RC Servo library using HW PWM at moment, along with my paying work.

lheusser, 2020-06-30:
Thanks @techpaul and @jaenrig-ifx for your further analysis!

> @lheusser I guess removing/commenting that line on the ACAN2517FD library could just do the work and have no impact on the SPI operation... You can give it a try.

 @jaenrig-ifx Commenting the line solves the compilation error. I will continue the integration of the MCP2518FD module and will be able to check if the SPI interface is working properly.

jaenrig-ifx, 2020-06-30:
Added usingInterrupt in pull request #152

closed 2022-01-06 by jaenrig-ifx

wzqvip, 2023-05-07:
Is it OK that we just comment the line "using Interrupt" ?