Infineon/XMC-for-Arduino #157: Assistance - PCLK calibration and tuning

techpaul, 2020-07-09:
Recent tests on preparing a Servo library brought up an issue with PCLK instability see comments on #155 

The basic issue over a batch of devices (mainly XMC1300 but will affect others) the PCLK was calculated to be around 0.7% out so PCLK instead of being 64 MHz was actually was worst seen case 64.470912 MHz, this may not seem much but this means many timed aspects from measuring time for _delay()_, _delaymicrosends)_ will have time drift, PWM duty cycles are out and other potential aspects.

As the same clock circuitry is used across XMC1 series and probably XMC4 series, the on-chip resonator and PLL are obviously locking at wrong frequency and can be seen to be 'hunting' on PWM outputs. To the extent I have seen a 50Hz timebase for a PWM wondering small amounts almost cycle to cycle. Wandering around the 50.34 Hz to 50.45 Hz

There are facilities for fractional div of PCLK generation to get different frequencies but this could potentially vary from device to device and across batches.

Similarly there is a clock adjustment for die temperature, which I am investigating but will involve a complex background task to operate and work out what will always work.

So quetion is has anybody else spent time getting the PCLK and MCLK timings adjusted or had experience to call upon to reduce research time ??

techpaul, 2020-07-12:
For those willing help who have ability to get develop branch after PR #158 and following equipment

- domestic freezer (or better) you may need household approval for this ...
- oscilloscope frequency counter 
- xmc board or many types
- ability to put laptop/computer next to freezer
- digital thermometer with wire to attach to PCB
- run PCLKtest.ino in tests folder

I am looking for tests in range +20C to -20C of PCB TEMP to monitor
- PCB Temp
- Die temp (on laptop)
- PWM 10kHz output ACTUAL measured frequency

Test is 
1. start outside freezer note readings
2. place in freezer
3. Monitor PCB temp and when it reaches a 5C step point note all readings
4. Once -20C has been achieved remove from freezer
5. Note readings again at 5C until reaches +20C

I welcome test results noting which board and any details like chip date code to try and get some statistics for before PCLK operation modifictions, later I would like to repeat test as much as possible against modifications.

HMenger, 2020-07-14:
I found this, maybe it is helpfull

http://www.infineon.com/dgdl/Infineon-ApplicationNote_AP32321_Oscillator_Handling-AN-v01_00-EN.pdf?fileId=5546d46253f65057015471bc512f774b

techpaul, 2020-07-15:
> I found this, maybe it is helpfull
> 
> http://www.infineon.com/dgdl/Infineon-ApplicationNote_AP32321_Oscillator_Handling-AN-v01_00-EN.pdf?fileId=5546d46253f65057015471bc512f774b

I have lready gone through that about 6 times, that has two parts

1 - Fractional Division of PCLK
Slightly tuning PCLK when PCLK is 32 MHz or less, this CANNOT be done when PCLK is 64MHz, and even at 32MHz I am not sure we can pull in the oscillator and lock the amount of variation I have seen.

 The steps available requires Calibration to an EXTERNAL reference that most people may not have. This sort of tuning is likely to be individual board specific and some boards may not be tunable. What the spread of that is related to temperature and ageing is not known. This is a do once and **HOPE** it does not vary over time or temperature.

How to create individual board calibration factors for fractional division, store the results and be able to incorporate them is not a simple task. With no idea of manufacturing spread it is impossible to determine optimum start point.

I did ask in a message that may have disappeared, for any manufacturing spread details for the oscillators, below is an example of what a manufacturing spread looks like, taken from an operational amplifier datasheet.
[image]

2 - Temperature adjustment on the fly
At the moment it is not known how much of PCLK inaccuracy is temperature, manufacturing or other factors, however the current process which is **NEVER** quick is to experiment and rewrite the poorly structured code from that application note into some code to automatically do temperature adjustment. I have done several times with many devices getting temperature spread of clocks tighter on many products and know this takes time to evaluate and needs to be done on many devices to be of any value.

The first stage of this of getting what exists as spreads BEFORE the change isi important along with changing code so Die temperature can be read as positive and negative values

closed 2022-01-06 by jaenrig-ifx