Infineon/XMC-for-Arduino #162: [XMC1100 H-Bridge 2Go] Serial Monitor is always empty

TonyCornuaud, 2020-08-18:
In <MotorControl-PWM.ino> source code, the Serial.println( "***" ) instructions seem not transmit data on USB port (i.e. COM Serial Port). Can you confirm?

techpaul, 2020-08-18:
Personally last tested it worked, do not have time to checks at moment hopefully someone else can before next week. Very busy till then.

TonyCornuaud, 2020-08-19:
Additional information...

I used the following configuration for my trial:

- Operating System: Windows 10 (same result w/ Windows 7)

- Arduino IDE: 1.8.13

- Arduino Board Manager: Infineon's XMC Microcontroller 1.5.0

- JLink Library Version: V6.82d

Can you give me your operating configuration to test it from my side?

Thank you,
Tony.

9Volts9er, 2020-08-24:
I just tried this and can confirm the issue. 
For testing I just used a simple sketch printing "Test" every 500ms.
The serial output works when selecting XMC2Go as board. But it does not work when H-Bridge 2go is selected.

Downgrading the XMC-for-Arduino version to 1.4.0 helps. Then the serial monitor prints "Test" again, even when the H-Bridge is selected as board. That means the issue came with some changes in between release 1.4.0 and 1.5.0.

I will dig into this further as soon as possible.

techpaul, 2020-08-24:
Still very busy but I suggest comparing pins_arduino.h for Hbridge as the first point to check.

9Volts9er, 2020-08-24:
Found the problem in the boards.txt:
-DSERIAL_ONBOARD was selected nevertheless which option for serial output is selected in the IDE.
I changed it to -DSERIAL_HOSTPC and now it works. The changes are already online with the latest commit.
As this was such a minor change, I didn't create a new pull request.

@TonyCornuaud can you please verify this, so this issue can be closed?

TonyCornuaud, 2020-08-25:
I just made the modification on the wrong line i.e:

- File Name:

> D:\users\cornuaud\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.5.0\boards.txt

- Wrong Line: 

> XMC1100_H_BRIDGE2GO.menu.UART.debug.uart.selected=-_DSERIAL_ONBOARD_

- To be repalced by following Line: 

> XMC1100_H_BRIDGE2GO.menu.UART.debug.uart.selected=-**DSERIAL_HOSTPC**

And now I have the text information on the serial port of the host machine.
The problem is fixed :-)

Thanks for your help,
Tony.

closed 2020-08-25 by TonyCornuaud