Infineon/XMC-for-Arduino #163: FR: Add support for KIT_XMC1400_ARDUINO

bobc, 2020-10-01:
This board appears to be similar to KIT_XMC14_BOOT_001 (which is now discontinued?), but the pin mapping is different.

techpaul, 2020-10-01:
Yes they are different boards, there is a branch for developing support for it.

Not sure when others are going to work on it when I get time I will put a few known things together for it as a starting point.

Also need to get a board myself.

bobc, 2020-10-01:
I am expecting to get a board soon. I have worked on several Arduino variants, maybe I can help. I am guessing the "develop" branch is the one to look at?

techpaul, 2020-10-05:
See Wiki for general details https://github.com/Infineon/XMC-for-Arduino/wiki/Contrubuting-and-Third-Party-Libraries

Also there is a note there for OLD XMC1400 Boot KIt (files in package but slectable)

Use XMC1100 Boot kit as starting point (with reference to old XMC1400 boot kit) as

- Same number of pins
- Has SAME requirement to have a RESET pin that is software driven, requires core flags changes)

Some chip port pins are differently assigned so needs careful checks.

techpaul, 2020-10-05:
See Wiki for general details https://github.com/Infineon/XMC-for-Arduino/wiki/Contrubuting-and-Third-Party-Libraries

Also there is a note there for OLD XMC1400 Boot Kit (files in package but NOT selectable)

Use XMC1100 Boot kit as starting point (with reference to old XMC1400 boot kit) as

    Same number of pins
    Has SAME requirement to have a RESET pin that is software driven, (requires core flags changes)

Some chip port pins are differently assigned so needs careful checks.

bobc, 2020-10-08:
Thanks for the info. I have got my board recently, so I will start in next few days.

closed 2021-07-19 by techpaul