Infineon/XMC-for-Arduino #164: Analog pin seems not to be an input

BrixInMotion, 2020-10-05:
I'm having problems reading the analog pins on an XMC4700. 

A. XMC4700 Relax Kit for 5V Shields
B. Lib Version 1.5.0
C. Arduino IDE Version 1.8.13
D. No extra lib, Hardware: 100k Potentiometer

All analog pins behave the same: When I ramp up the voltage with the potentiometer everything works fine until 1V. Afterwards it looks like in the picture below. 
[image]

According to the picture, the Serial output prints random values between 0 and 1023.
For me it looks like the pin is defined as output and is not floating. I'm using minimal code.

```
uint16_t result = 0;  
void setup() {  
   Serial.begin(115200);  
   wiring_analog_init();  
}  
void loop() {  
  result = analogRead(A5);  
  Serial.println(result);  
  delay(500);  
}
```

techpaul, 2020-10-05:
Random values and seeing 25 MHz oscillations, normally means BAD ground (0V) or other connections. It may not be but that is majority of this type of issues.

Code note you do NOT need the wiring_analog_init() call in your code it is ALWAYS done in the core before setup is called.

Going back through datasheets, circuits etc, I can STATE that analog inputs especially A0 to A21 are PHYSICALLY INPUTS only and cannot physically be configured for ouputs. With exception of A13 and A14 which can be made ONLY to DAC outputs.

I can see nothing YET to suggest the problem is in the software. If I get some time I will run up a board and duplicate this.

BrixInMotion, 2020-10-05:
Thanks for the fast response. I would eliminate that it's a Hardware problem as I use exactly the same wiring for an Arduino UNO where it works perfectly. The wiring_analog_init() was only for testing, but nice to know that I don't need it for sure.

techpaul, 2020-10-05:
Please NOTE I am a contributor and do not work for Infineon

I am sorry I have over 40 years experience in electronics, computers (from filling rooms to tiny almost wearable) and many many different manufacturers and chip types. Just because hardware worked on one system does not mean ALL connections and parameters are identical, i have seen it too many times especially with breadboards and flying wires. Check the SAME hardware still works again on UNO, could be faulty pot.

The only time i have seen oscillations in the past is mis-wiring, intermittent connections or missing grounds (even on scope probes). I have seen things like worn pot internal conductive tracks or even cracks in tracks.

There is a POSSIBILITY that your 100k pot is TOO large a value, I have not yet foun the input impednace for an Analog input, so your pot could be a mismatch. Many new processors have only 30 to 50k input impedance compared to old style using 1M Ohm or greater input impedance. So when your pot reaches 1V level the impedance is roughly 20% of full travel (20k Ohm) and the impedance mismatch along with stray capacitance (or broken ground) is causing inaccuracy AND oscillations.

First try a 10k or 1k pot instead.

My personal candidates are

1.  Broken pot
2.  Wiring issue
3.  Impedance mismatch
4.  other hardware problem.

These types of issues are normally (99.9% of the time) hardware issues.

@mhollfelder and @jaenrig-ifx can anyone find any definitive details on the hardware Analog input impedance in VADC mode for an analog input, this is NOT in any datasheet for XMC4700 or any XMC4 series I can find

BrixInMotion, 2020-10-06:
Ok so I just did some exact testing. I used a new 1K linear poti, double and tribble checked the wires and used even a second, unopened XMC4700 kit to check. It behaves exactly as described above but ONLY on pins A4 and A5, so I'm sorry about what I've written before, Pins A0- A3 are normal as they should behave.
The Poti is connected between 3,3V and Gnd. The exact thresholds, between the noise appears are above 1,6V and below 2,6V.
I don't think that it has something to do with impedance. The poti just works as a voltage divider and provides an absolute voltage on the analog pin. So there should be no difference in using a 100k or 1k poti.

techpaul, 2020-10-06:
Right so we know A0 to A3 are fine A4 and A5 have an issue which MIGHT be a software setup issue, that needs checking as this may be what groups are used in VADC.

What is the performance on A6 to A21 ?

For analogue AND digital circuits, input and output impedance is IMPORTANT. in Digital circuits if a gate or output can drive 10 input loads that is because the parallel input impedance of 10 input loads is at **LEAST** 100 x larger than the output impedance of the gate or output. This is simple fact borne out of the principles of Optimised Power Transfer, 
circuit loading. and simple parallel impedance networks, output and inputs are one COMBINED circuit. For good digital and analogue circuits to avoid loading output impedances are VERY low (< 100 Ohm), input impedances are often measured in the Mega, Giga or some cases Terra Ohms. If you had a Voltage Follower circuit between the pot and the ADC your statement would be true. This sort of circuit is often an op-amp in Unity Gain configuration or NPN transistor 
in Common Collector configuration. With direct connection of Pot to ADC input it is ONLY true when the input impedance of the ADC is at **LEAST** 100 times larger than the output impedance of the circuit driving it. This is basic electronics.

To model the circuit you have a pot which is equivalent to TWO series resistors, with the one tied to 0V in parallel with the ADC input also going to 0V. Lets call the two pot resistors R1 (tied to supply) and R2 (tied to 0V, and the ADC input impedance Zin, which means Zin is in parallel with R2, and R1 is feeding the parallel combination. So as R1 increases R2 decreases in a matching ratio, also assume the pot has a linear resistance track.

You are setting the voltage by reading a meter or scope on the pot, but with different Zin or total Pot resistance the angular position to achieve the same voltage out is DIFFERENT. You will also note in regions it is easier or harder to achieve small incremental step changes (e.g. 1.25V to 1.26V) The only thing you can say about pots of different values, is the ratio of the resistances of the POT, should be the same for degrees of rotation. The parallel effect of Zin and R2 will be DIFFERENT depending on the value of the pot. Also the same value pot use on two different ADCs with different Zin **WILL** be different voltage relationship to position of pot rotation.

So if we use your example of 100k pot and for easier calculations the pot has 250 degrees of rotation between ends, then that gives 400 Ohms per degree of rotation. Inversely that is 2.5 degrees per kOhm. So calculating the data for a Zin of the ADC being 30k or 1M gives different responses related to degrees of rotation. See the graph below

[image]

Impedance mismatch on ADCs is a common issue I see mainly on newer micros from ANY manufacturer, and is often overlooked by those in a hurry or inexperienced. This will cause wrong readings or apparent incorrect response or accuracy over range.

techpaul, 2020-10-06:
All A0-A21 is supposed to be support for XMC4700 boards extract from pins_arduino.h

~~~
    /* 17  */   {XMC_GPIO_PORT14 , 0},  // A0 / ADC Input               P14.0 (INPUT ONLY)
    /* 18  */   {XMC_GPIO_PORT14 , 1},  // A1 / ADC Input               P14.1 (INPUT ONLY)
    /* 19  */   {XMC_GPIO_PORT14 , 2},  // A2 / ADC Input               P14.2 (INPUT ONLY)
    /* 20  */   {XMC_GPIO_PORT14 , 3},  // A3 / ADC Input               P14.3 (INPUT ONLY)
    /* 21  */   {XMC_GPIO_PORT14 , 4},  // A4 / ADC Input               P14.4 (INPUT ONLY)
    /* 22  */   {XMC_GPIO_PORT14 , 5},  // A5 / ADC Input               P14.5 (INPUT ONLY)
    /* 23  */   {XMC_GPIO_PORT3 , 12},  // SPI_SS_2                     P3.12
    /* 24  */   {XMC_GPIO_PORT5 , 9},   // Additional LED1              P5.9
    /* 25  */   {XMC_GPIO_PORT5 , 8},   // Additional LED2              P5.8
    /* 26  */   {XMC_GPIO_PORT15 , 13}, // Additional BUTTON1           P15.13 (INPUT ONLY)
    /* 27  */   {XMC_GPIO_PORT15 , 12}, // Additional BUTTON2           P15.12 (INPUT ONLY)
    /* 28  */   {XMC_GPIO_PORT4 , 1},   // SPI_SS_3 (SD CARD)           P4.1
    /* 29  */   {XMC_GPIO_PORT3 , 5},   // SPI-MOSI (SD CARD)           P3.5
    /* 30  */   {XMC_GPIO_PORT4 , 0},   // SPI-MISO (SD CARD)           P4.0
    /* 31  */   {XMC_GPIO_PORT3 , 6},   // SPI-SCK  (SD CARD)           P3.6
    /* 32  */   {XMC_GPIO_PORT1 , 6},   //                              P1.6
    /* 33  */   {XMC_GPIO_PORT1 , 7},   //                              P1.7
    /* 34  */   {XMC_GPIO_PORT1 , 4},   // USB Debug RX                 P1.4
    /* 35  */   {XMC_GPIO_PORT1 , 5},   // USB Debug TX                 P1.5

	//Additional pins for port X1 starting here
	/* 36  */   {XMC_GPIO_PORT3 , 4},   // PWM42-2                      P3.4
	/* 37  */   {XMC_GPIO_PORT0 , 5},   // I2C_1 SDA  // SPI_4 MOSI     P0.5
	/* 38  */   {XMC_GPIO_PORT0 , 3},   // PWM80-2                      P0.3
	/* 39  */   {XMC_GPIO_PORT0 , 1},   // PWM80-1                      P0.1
	/* 40  */   {XMC_GPIO_PORT0 , 10},  //                              P0.10
	/* 41  */   {XMC_GPIO_PORT3 , 2},   //                              P3.2
	/* 42  */   {XMC_GPIO_PORT3 , 1},   //                              P3.1
	/* 43  */   {XMC_GPIO_PORT15 , 6},  // A6 / ADC Input               P15.6 (INPUT ONLY)
	/* 44  */   {XMC_GPIO_PORT15 , 5},  // A7 / ADC Input               P15.5 (INPUT ONLY)
	/* 45  */   {XMC_GPIO_PORT15 , 3},  // A8 / ADC Input               P15.3 (INPUT ONLY)
	/* 46  */   {XMC_GPIO_PORT14 , 15}, // A9 / ADC Input               P14.15 (INPUT ONLY)
	/* 47  */   {XMC_GPIO_PORT14 , 13}, // A10 / ADC Input              P14.13 (INPUT ONLY)
	/* 48  */   {XMC_GPIO_PORT14 , 7},  // A11 / ADC Input              P14.7 (INPUT ONLY)
	/* 49  */   {XMC_GPIO_PORT15 , 15}, // A12 / ADC Input              P15.15 (INPUT ONLY)
	/* 50  */   {XMC_GPIO_PORT14 , 9},  // DAC0 // A13 / ADC Input      P14.9 (INPUT ONLY)
	/* 51  */   {XMC_GPIO_PORT2 , 13},  //                              P2.13
	/* 52  */   {XMC_GPIO_PORT5 , 10},  //                              P5.10
	/* 53  */   {XMC_GPIO_PORT5 , 11},  // PWM80-0                      P5.11
	/* 54  */   {XMC_GPIO_PORT1 , 14},  //                              P1.14
	/* 55  */   {XMC_GPIO_PORT14 , 8},  // DAC1 // A14 / ADC Input      P14.8 (INPUT ONLY)
	/* 56  */   {XMC_GPIO_PORT15 , 14}, // A15 / ADC Input              P15.14 (INPUT ONLY)
	/* 57  */   {XMC_GPIO_PORT14 , 6},  // A16 / ADC Input              P14.6 (INPUT ONLY)
	/* 58  */   {XMC_GPIO_PORT14 , 12}, // A17 / ADC Input              P14.12 (INPUT ONLY)
	/* 59  */   {XMC_GPIO_PORT14 , 14}, // A18 / ADC Input              P14.14 (INPUT ONLY)
	/* 60  */   {XMC_GPIO_PORT15 , 2},  // A19 / ADC Input              P15.2 (INPUT ONLY)
	/* 61  */   {XMC_GPIO_PORT15 , 4},  // A20 / ADC Input              P15.4 (INPUT ONLY)
	/* 62  */   {XMC_GPIO_PORT15 , 7},  // A21 / ADC Input              P15.7 (INPUT ONLY)
~~

The holes exist for the extra channels on the multiway connector mountings and are detailed in the schematic for the board. When I get some time XMC4700 pin out needs second page for the extended pins. Recently did a minor update for XMC4400 pinout  diagram.

I suspect the issue IF software will be in XMCLIB and more than likely VADC group related, so a scan of which ones work and which ones don't will help give data on what is actually working. This may be a hardware related issue where someone has not documented someting properly, having just found one on a different board.

BrixInMotion, 2020-10-07:
All right, I tested all analog pins and can say for sure that the only faulty pins on my both XMC4700 5V kits are A4 and A5. A0 to A3 and A6 to A21 work fine.
I tested a Lite Kit as well, interestingly no pin there seems faulty. So I will check either other 5V kits ore the PCB design, if there are differences between the lite and the 5V. As this is no topic concerning the XMCforArduino Lib anymore, I will close this issue. Thanks for your support @techpaul !

techpaul, 2020-10-07:
Glad this is sort of sorted.

Contact again if need assistance

BrixInMotion, 2020-10-22:
We found the reason for this issue: there are two traces too many on the board, that connect P14.5 (A5) with P0.13(SCL) and P14.4(A4) with P3.15(SDA). This is a unnecessary limitation of the board, as the XMC4700 would have separate pins for ADC and I2C unlike the ATmega328. 
A solution here, in order to use both analog pins and I2C is to take a knife and carefully cut the two traces near R104 and R105 shown in the picture .

[image]

mhollfelder, 2020-10-22:
Hi BrixInMotion,

This connection is in place as for the Arduino Uno Rev3 pin-out, these pins are connected and shared I2C/Analog pins as well:

https://content.arduino.cc/assets/UNO-TH_Rev3e_sch.pdf

The I2C pins on the top are connected to the pins on the bottom of the board itself.
You can also see this from the pin-out as they are using the same signals:

https://content.arduino.cc/assets/Pinout-UNOrev3_latest.png

As some shields rely on this connection, this is implemented here by having this on-board pin connection.
You can either cut it or you set the I2C pins to tri-state to not influence the analog measurements (which blocks using the I2C, of course).

I hope I could give some background why this connection is in place here.

Thanks and best regards,

Manuel

techpaul, 2020-10-22:
> Hi BrixInMotion,
> 
> This connection is in place as for the Arduino Uno Rev3 pin-out, these pins are connected and shared I2C/Analog pins as well:
> 
> https://content.arduino.cc/assets/UNO-TH_Rev3e_sch.pdf
> 

This is where there is a Yes and No answer, or as is sometimes said "Comparing Oranges and Apples".

On AVR UNO this is like some of the XMC1xxx boards an INTERNAL alternate pin configuration, on the XMC4700 it is linking pins externally, and at best should have been onboard jumper or individual 0R resistors that could be removed.
Otherwise we are saying you cannot have these two analog inputs (see later). Not checked whether XMC1400 Arduino Kit is Alternate pin function or pins externally connected. This has OTHER issues **NOT DOCUMENTED**.

> The I2C pins on the top are connected to the pins on the bottom of the board itself.
> You can also see this from the pin-out as they are using the same signals: [...]

Setting pins to Tri-state is **NOT DIRECTLY** supported in XMC-for-Arduino, setting as input will add a load and leakage to the analogue signal.

The **NEAREST** setting is setting the pins to OUTPUT_OPENDRAIN and writing a 1 to the port pin, to turn output off.
The length of tracks may well add a lot more noise to analogue readings, as well as any noise leakage from pin(s).

The files like pins_arduino.h do **NOT** properly document this, nor does

- Pin out diagram(s) (including XMC1400 Arduino Kit, XMC4400, possibly others)
- Wiki
- Similarly AREF pin is NOT properly documented best use is leave OPEN do not touch

No mention is made of fact that you cannot use I2C and these analogue inputs at the same time in any part of the documents or Wiki.

This needs a documentation overview **WITH** notes on track cuts where pins are externally connected as well as how to set I2C/GPIO pins to avoid interacting with analogue pins.

This artefact does **NOT** affect the following boards at current revisions

- Any of XMC1100 boards (XMC1100 Boot Kit is the same format)
- XMC1300 Boot Kit

This artefact **DOES** affect

- XMC4700 boards
- XMC4400 Platform 2 Go
- AMC1400 Arduino Kit (XMC1400 Boot kit is discontinued and not checked)

mhollfelder, 2020-10-22:
I agree to techpaul's comments, ideally it would be an alternate multiplexed pin function which is properly documented.
For the Relax Kit, I cannot fully judge what was the reasoning behind making such a choice of connection.

I can only comment on why the connection is in place, not why the connection was chosen as it is. 
Ideally, of course you should have a 0 Ohm resistor or a switch as the best solution.

XMC1400 Arduino Kit has the SAME pins P2.0 and P2.1 connected on the board between I2C and ADC pins.
So it is an alternative pin function and I2C and ADC functions can be activated via these functions.

Tristate can be set via `pinMode(<pinNumber>, XMC_GPIO_MODE_INPUT_TRISTATE);`, the regular definitions can be used here.
Of course, this has to be documented properly, better would be to include the tristate as an optional mode if required.

Fully agree, it has to be properly documented as the only source where you can find it are the user manuals.

I added two chapters to the XMC1100 Boot Kit and XMC4700 Relax Kit Wiki for documentation.
I will check for the XMC4400 Platform 2 Go and also document it.

Thanks for investigating this and sorry that it caused such troubles.

techpaul, 2020-10-22:
Thanks for putting an update on the XMC4400 and XMC4700 pages
1.  The XMC4700 page shows the XMC4400 image for me which is different tracks to cut
2.  XMC4400 page shows old Pinout diagram there is a PR on Assets to correct a couple of issues, but really needs  additional modifications
3.  See no difference on XMC1100 Boot Kit page (they are actually separated on that board.

Can I suggest also showing OPENDRAIN method as well as the tristate method is only really usable on known very low impedance analogue sources such as op-amps. I have seen that tristate outputs generally leak around 1-10 microamps which can affect analogue outputs from source impedances greater than 100R especially on end of ADC ranges.

As soon as I can will look at updating pinout diagrams and potentially adding more pinout sheets for XMC4700 and XMC4400 beyond the UNO size shield connections. I will t.ry and get an update to analogue functions pages later tonight

mhollfelder, 2020-10-23:
> 1. The XMC4700 page shows the XMC4400 image for me which is different tracks to cut

I think I have overwritten the changes and my browser hasn't shown it as it was cached - sorry for that!
In the page history, there is an earlier [version](https://github.com/Infineon/XMC-for-Arduino/wiki/XMC4700-Relax-Kit/f6da76e96130393769aab1b8ae5599e058066c73) which has the right picture.
I fixed it and updated the page accordingly.

> 2. XMC4400 page shows old Pinout diagram there is a PR on Assets to correct a couple of issues, but really needs additional modifications

Has to be fixed - I also saw it, but had no time to look into it.
I will check it when I have time and update it accordingly.

> 3. See no difference on XMC1100 Boot Kit page (they are actually separated on that board.

I just added a note here:

https://github.com/Infineon/XMC-for-Arduino/wiki/XMC1100-Boot-Kit#important-note

It is supposed to clarify the differences to other boards and the official one.

BrixInMotion, 2020-10-23:
Before providing the solution with cutting the traces, we already considered changing the output/input state of P0.13 / P3.15. So you're right, the ADC Pins work fine, when you set both I2C pins to Input-Tristate, but **ONLY** on boards without level shifter, like the XMC4700 Relax lite kit.
I've got the strong presumption, that the [level shifter](https://www.ti.com/lit/ds/symlink/txs0108e.pdf) is causing the noise I recorded (the oszi screenshot at the beginning). So in order to use the two analog pins on the 5V kit, you have to cut the traces **or** disable the level shifter. This can be done by soldering a 0 Ohm resistor bridge in place of R301, so the OE pin is tied to ground (I can verify it works as I just tested it).

techpaul, 2020-10-23:
> Before providing the solution with cutting the traces, we already considered changing the output/input state of P0.13 / P3.15. So you're right, the ADC Pins work fine, when you set both I2C pins to [...]

This in some ways does NOT surprise me, setting the pin tristate or open drain is making that side of the level converter floating (ish) with leakage and track acting as aerial picking up noise, and probably proximity of people. The noise is supply noise transmitted through IOREF pin and its internal pullups to output pin.

To use analogue pin cutting that track CLOSE to analogue input is only choice in that configuration

techpaul, 2020-10-23:
> I think I have overwritten the changes and my browser hasn't shown it as it was cached - sorry for that!
> In the page history, there is an earlier [...]

I am in process of updating pinout diagrams and noted there is a MISTAKE in XMC4700 pin out as it numbers A0 to A5 in reverse order to other boards and pins_arduino.h

Later today I will put a PR into Assets with updated details

> > 1. XMC4400 page shows old Pinout diagram there is a PR on Assets to correct a couple of issues, but really needs additional modifications
> 
> Has to be fixed - I also saw it, but had no time to look into it.
> I will check it when I have time and update it accordingly.

Updated pinout in progress to add
1.   analogue and I2C  pin
2.  Uno R3 compatibility note
3.  Add weblink for Wiki

This is being done for XMC4400/XMC4700/XMC1100

Await updates later for this and more comments in pins_arduino.h for boards

techpaul, 2020-10-23:
Updated pinout diagrams as PR 6 on Assets repository comment is

**Up Rev Pin outs to V1.1.0**

Boards XMC1100 Boot Kit, XMC4400 Platform 2Go, XMC4700 Relax

Update Shared I2C and ADC pin status, Uno R3 comparability notes
Correct XMC4700 A0 to A5 Arduino pin order
Add wiki URL to diagram

**WARNING renames XMC4400 Platform 2Go filename to be in line with other pin out diagrams to end in "_PO.jpg"**

When this is done can edit wiki and add a couple of items to match

techpaul, 2020-10-25:
Added another page on Wiki for I2C and analog pins for all boards

https://github.com/Infineon/XMC-for-Arduino/wiki/I2C-Analog-pins

@mhollfelder when you get a moment the revisied pin out diagrams are awaiting in Assets repo

mhollfelder, 2020-10-25:
> @mhollfelder when you get a moment the revisied pin out diagrams are awaiting in Assets repo

Done and merged plus added some minor modifications.

techpaul, 2020-10-26:
Thanks for getting some updates through, at least this is documented.

Need to get time for worling on a few other issues

closed 2021-06-29 by techpaul