Infineon/XMC-for-Arduino #167: Maximum flash size for XMC1300 Sense2GoL is incorrect

yennster, 2021-01-06:
According to the boards.txt page the Sense2GoL has `125160` bytes flash: https://github.com/Infineon/XMC-for-Arduino/blob/master/boards.txt#L186

But Block0 on the Sense2GoL (https://www.infineon.com/dgdl/Infineon-Sense2GoL_development_kit-PB-v01_00-EN.pdf?fileId=5546d462636cc8fb0163eeb7b2d06f35) actually only has `32768` bytes of flash.

techpaul, 2021-01-11:
@mhollfelder @yennster

This actually has a series of issues relating to Sense2GoL and XMC1300 Boot Kit for linker scripts and boards file.
* Linker script is for T038X0200 or Q040X0200 (200 kB of Flash) as in XMC1300 Boot Kit
* Sense2GoL uses XMC1302T016X0032ABXUMA1 variant T016X0032 according to data sheet

This means boards.txt is going to have to refer to TWO different linker_script.ld files possibly in config folders for the two different boards also other settings will have to be changed in particular but not limited to

*  XMC1300 Boot Kit
  * XMC1300_Boot_Kit.upload.maximum_size=65536 --> 204800
  * XMC1300_Boot_Kit.build.flash_size=64K  --> 200K
  * XMC1300_Boot_Kit.build.flash_ld=linker_script.ld  -> config/\XMC1300_Boot_Kit/linker_script.ld
  * move existing linker_script.ld to config/\XMC1300_Boot_Kit
* XMC1300 Sense2GoL
  * XMC1300_Sense2GoL.upload.maximum_size=125160 --> 32768
  * XMC1300_Sense2GoL.build.board.type=T038x0064  --> T016x0032
  * XMC1300_Sense2GoL.build.flash_size=64K   --> 32K
  * XMC1300_Sense2GoL.build.flash_ld=linker_script.ld --> config/XMC1300_Sense2GoL/linker_script.ld
  * Copy existing linker_script.ld --> config/XMC1300_Sense2GoL
  * edit and test this new linker_script.ld for T016x0032

This needs testing but I **personally** only have XMC1300 Boot Kit to hand

techpaul, 2021-01-11:
@mhollfelder @yennster

On reflection I will work on making several linker_script files for different flsh sizes to avoid paths and so other boards can be added calling the relevant file meaning less duplication of files.

The ONLY thing that changes is the Flash size for linking so different versions will be made for

* linker_script_32k.ld
* linker_script_64k.ld
* linker_script_128k.ld
* linker_script_200k.ld

Suitable changes will be made to bords.txt nd other files, and should be uploaded later today into develop, you should be able to just download from develop branch boards.txt and the linker scripts to get things correct.

techpaul, 2021-01-11:
Taking longer than expected as found other errors in pins_arduino.h which thinks the CPU is 32 pin when it is 16pin

techpaul, 2021-01-12:
Develop branch has updated files to correct the flash sizes for all types of XMC1300 see Pull request #170 

Please test on actual software

yennster, 2021-01-12:
@techpaul You are awesome. Will give it a go this afternoon

closed 2021-06-29 by techpaul