Infineon/XMC-for-Arduino #168: getTemperature() reports 4294967087 degC

schumar, 2021-01-10:
> Which XMC board are you using

XMC 2Go-V1

> Which version of XMC-for-Arduino are you using

1.5.0

> Which version of Arduino IDE and platform you are using

1.8.13

> Does this happen in an example programme or empty sketch

Used the `DieTemperatureMeasurement` example, as well as my own program

> Example minimal code or links to code that demonstrate the issue

[libraries/DeviceControlXMC/examples/DieTemperatureMeasurement/DieTemperatureMeasurement.ino](https://github.com/Infineon/XMC-for-Arduino/blob/master/libraries/DeviceControlXMC/examples/DieTemperatureMeasurement/DieTemperatureMeasurement.ino)

Output:
```
Die Temperature	4294967087
Die Temperature	4294967087
Die Temperature	4294967087
[...]
```
Converting this `uint32_t` to `int32_t` gives `-209` degC, so it looks like `XMC_SCU_CalcTemperature()` is returning `64` Kelvin (`0x40`), see `libraries/DeviceControlXMC/src/DeviceControlXMC.cpp`:
```c++
/**
 *  Get Die temperature in Celsius
 * @return temp_celsius: Die temperature in Celsius
 */
uint32_t XMCClass::getTemperature( )
{
        uint32_t temp_calc = 0, temp_celsius = 0;
        temp_measurement_s = XMC_SCU_STATUS_ERROR;

#if (UC_FAMILY == XMC4)
        // [...]
#elif(UC_FAMILY == XMC1)
        while( temp_calc == 0 )
         {
                 XMC_SCU_StartTempMeasurement( );
                 temp_calc = XMC_SCU_CalcTemperature( );
         }
        temp_celsius = temp_calc - 273;
#endif
        return temp_celsius;
}
```

techpaul, 2021-01-10:
So I got out
* XMC1100 2GO
* Run up Arduino IDE V1.8.9
* Select board and port for the 2GO board
* Open Example for Die temp measurement
* Open Serial Monitor
* Compile and upload to board 
* after a few readings use Freezer spray on board and got results below

~~~
XMC Die Temperature in Degree C Demo  (V2.0)
Running on XMC 1100 XMC2GO
Die Temperature	36
Die Temperature	36
Die Temperature	36
Die Temperature	36
Die Temperature	36
Die Temperature	27
Die Temperature	25
Die Temperature	13
Die Temperature	-13
Die Temperature	-18
Die Temperature	-21
Die Temperature	-25
Die Temperature	-26
Die Temperature	-25
Die Temperature	-24
Die Temperature	-23
Die Temperature	-21
Die Temperature	-20
Die Temperature	-19
Die Temperature	-16
Die Temperature	-15
Die Temperature	-14
Die Temperature	-13
Die Temperature	-12
Die Temperature	-11
~~~

So with NO changes to standard example or libraries, got sensible results

techpaul, 2021-01-10:
Your copy of Devicecontrol needs updating from develop branch I modified for negative temps in July 2020 to allow negative temperatures. We need a version update.

closed 2021-06-29 by techpaul