Infineon/XMC-for-Arduino #172: Infineon DemoSense2Go XMC4700 programs don't work

ashwinj92, 2021-04-21:
Hi Team,

The examples for DemoSense2Go XMC4700 base board do not work and the Led colors dont change at all. 
I get the following warning log when I compile the sketch. Any solutions would help.

WARNING: library BGT24LTR11-Pulsed-Doppler-Radar claims to run on cortex-m4 architecture(s) and may be incompatible with your current board which runs on arm architecture(s)

After following the tutorial in this documentation: https://www.infineon.com/dgdl/Infineon-AN605_Sense2GoL_Pulse_Demo_Board-ApplicationNotes-v01_00-EN.pdf?fileId=5546d462700c0ae601708bd8d71f1440

The Leds do not change colors and remain constant green after programming it. Not sure what is the issue with the board.

mkiessling-ifx, 2021-04-22:
Dear Ashwinj92, 

I assume you already stuck with the most simplest example, right? 

please have a look at the examples and libraries provided in the branch 
https://github.com/Infineon/XMC-for-Arduino/tree/24Ghz-radar

Best Regards
Michael

techpaul, 2021-04-22:
The problem is you have OLD version of library.

In a library folder is a file called library.properties that has an INCORRECT architecture value. The architure in this file is teh Arduino Board type support NOT processor, should really say 
~~~
architectures=xmc,arm
~~~
But this is sufficient (as in current release)
~~~
architectures=arm
~~~

Current version of library has this correct.

I suggest you update the Boards support form Arduino IDE

Tools -> Board....... -> Board Manager

(See installation notes)

ashwinj92, 2021-04-27:
Thank you for the replies, I have updated the board libraries to have the latest version of the library installed on the Arduino IDE as well. I'm trying to extract raw signals (I and Q for further processing) the serial monitor does not show any data at the specified baud rate. 

https://github.com/Infineon/XMC-for-Arduino/tree/24Ghz-radar and have used this documentation to the dot. And Also using the correct COM port aswell.

I get the following warning log when I compile the sketch. Any solutions would help.

WARNING: library BGT24LTR11-Pulsed-Doppler-Radar claims to run on cortex-m4 architecture(s) and may be incompatible with your current board which runs on arm architecture(s)

techpaul, 2021-04-27:
The radar branch is separate from the main release, I have tried some of the standard release V1.5.0 and EXAMPLES for Sense2Go and had NO compiler warnings for boards XMC 4400 Platform2Go or DEMO Radar BB XMC4700. 

The wrning you are seeing is an Arduino IDE compiler warning based on board setting and library.properties file in the libraries tree, having checked both main release and 24GHz Radar branch I cannot see any errors in those files.

Some examples do NOT work for ANY board, you must have the correct confifuration for board (Tools -> Board....)

Important questions

1/ What version of Arduino IDE do you hve
2/ How did you install the 24GHz branch
3/ What EXACT name of boards or board set do you have
4/ What is  your Arduino IDE board setting
5/ Which examples have you tried using

To prove you are setup on correct COM port compiling and getting output try compiling and uploading the example

Device Control XMC -> DieTemperatureMeasurement
or
Device Control XMC -> StackMemoryMeasurement

These will output data onto serial port and prove you are getting correct compilation for that board when output works

techpaul, 2021-07-19:
Items have moved to separate branch and projects by looks of things so closing

closed 2021-07-19 by techpaul

tinymanta, 2022-09-26:
@techpaul any updates on this topic?

techpaul, 2022-09-26:
Check the 2,4 GHz branch and see if Infineon have any further answers.  I am just a contibutor.