Infineon/XMC-for-Arduino #173: "size" of XMC_IFX_1.2.0.zip in package.json has trailing spaces

schumar, 2021-05-12:
Using `arduino-cli`, I get
```
$ arduino-cli core update-index
Error updating index: invalid package index in
https://github.com/Infineon/XMC-for-Arduino/releases/latest/download/package_infineon_index.json:
json: invalid number literal, trying to unmarshal "\"2074378  \"" into Number
```
because the `size` parameter has trailing spaces:
```
        {
          "name": "Infineon's XMC Microcontroller",
          "architecture": "arm",
          "version": "1.2.0",
          "category": "Contributed",
          "url": "https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.2.0/XMC_IFX_1.2.0.zip",
          "archiveFileName": "XMC_IFX_1.2.0.zip",
          "checksum": "SHA-256:b1b6e253e7062bd441e6d8572d16a7cba432c440676c4499caa825fd4cafbf3b",
          "size": "2074378  ",
[...]
```
(same for `XMC_IFX_1.1.2.zip`)

techpaul, 2021-06-22:
Sorry for any late response I have checked through package_infineon_index.json as CURRENTLY stands for release V1.5.0 and I see no such issue.

Can I suggest first getting V1.5.0 package_infineon_index.json from the releases section here and trying again.

This may be an old file issue or some other corruption.

techpaul, 2021-06-22:
Note newest version of package_infineon_index.json conatins older version details as well.

techpaul, 2021-07-19:
Doing reviews of automated release process see different issue, whilst I have write permissions for this repository I do not have that for Infineon/Assets so I have instigated a Pull Request to correct the master file. Apologies for any delay this is not my full time job and I have been very busy

schumar, 2021-07-19:
Thanks @techpaul , your PR should fix the problem!

techpaul, 2021-07-20:
If pull request  not actioned in a few weeks I will see if we can do another method

techpaul, 2021-07-22:
@schumar Things are slightly more complicated

Since V1.4.0 releases are done by automated scripts which has a fundamental ASSUMPTION all previous package_infineon_index.json are perfect.

So when the inefficient script runs it takes as source the last release file to merge the new release in. Files in release area are NOT changeable, there is no editable source to correct any form of omission or error or corruption. So I am between other jobs seeing if I can 
* get a clean version for you to point to for now
* Tidy up scripts and remove redundant actions and bloat
* Test on my repository (techpaul/XMC_for_Arduino)
* Merge to develop here
* Merge develop to master here
* Make a new release version

Most of this is extra work for me and my permissions level  **MIGHT NOT** allow me to perform a release

techpaul, 2021-07-24:
@schumar

Try pointing to this URL for a clean package_infineon_index.json for now

https://raw.githubusercontent.com/techpaul/XMC-for-Arduino/master/package/package_infineon_index.json

At least let me know if it works

jaenrig-ifx, 2021-09-10:
All "package_infineon_index.json" in this and _assets_ repo are now updated. Removed trailing spaces for size fields.
Solved as long as the release automation does not introduce this error again... which has not for the 1.5.0 release.

closed 2021-09-10 by jaenrig-ifx