Infineon/XMC-for-Arduino #175: Arduino IDE can't connect XMC_1100_2GO

hubertlxb, 2021-06-22:
I follow all the procedures to configure the Arduino.
However, arduino indicate that the XMC 1100 is not in the COM.
[image]
In additon, the J-link driver was installed,

techpaul, 2021-06-22:
Have you selected the correct COM port ?

COM3 seems surprisingly LOW number for a USB device..

On MY system the USB devices come up in the COM30 to COM50 range when i do

   Tools -->  Port 

I get for a XMC1000 2Go the list
COM1
COM2
COM47

hubertlxb, 2021-06-23:
Hi Paul,

thanks for your support!
I changed the Port number, but it still doesn't work.

[image]
[image]
[image]
[image]
[image]

techpaul, 2021-06-23:
This suggests that something is not connected properly or has broken connection to the XMC11002Go or on the board

I have no problems with multiple XMC2Go boards

Have you checked Wiki for 64 bit system issues?

Have you tried compiling and uploading a blank sketch?

Copy the TEXT not picture of error messages

hubertlxb, 2021-06-24:
tried compiling and uploading a blank sketch, but the error is still there.
following you can find the TEXT message.

Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Windows 10
Operating System: Windows 10

Initialisation
--------------------------
Windows 10

[Error] Infineon.DebuggerExceptions: It seems that JLink software is not installed please download from www.segger.com and install it. You can specify it by setting java property xmcFlasher.JLink.dllPath 
Error downloading https://github.com/Infineon/XMC-for-Arduino/releases/latest/download/package_infineon_index.json

techpaul, 2021-06-24:
See Issue #104 and also

https://github.com/techpaul/XMC-for-Arduino/wiki/Windows-10-64-bit-Flasher

closed 2022-01-06 by jaenrig-ifx