Infineon/XMC-for-Arduino #176: Call to Infineon team for new release

techpaul, 2021-07-18:
@Infineon @jaenrig-ifx @mhollfelder 

After someone has added extra layers of corporate bureaucracy to avoid being able to contact maintainers This has to be reported as an Issue, no I do not want to have yet ANOTHER unanswered forum on a different site to contend with AS WELL !!!

The XMC-for-Arduino project needs a new release of merging develop into master branch and changing the architecture from arm to xmc (with knock on effects to automated release scripts and travis-ci).  Last release was June 2020

**WHO IS THE CURRENT PERSON(S) RESPONSIBLE FOR RELEASES** 

As a contributor with some rights to this repository

Due to the fact that many parts of this repository rely on access to central repository Infineon/Assets (package_index.json with write permissions) and lack of documentation on the release process how to add new boards into build and release process, it is VERY DOUBTFUL I have sufficient permissions to actually perform a release.

Longer term this project requires
1. More admin documentation
2. Assets resources sources to be within this repository
3. Way to contact maintainerS (yes multiple) especially from known contributors

Whilst Infineon/Assets reporitory resources maybe helping your main site web devs it makes maintaining this repository difficult, the outputs should be pull request on that repository from this repository

ederjc, 2021-08-10:
Hello Techpaul,

Many thanks for your feedback. It helped us for the internal discussions that were already ongoing regarding the XMC for Arduino.
We are aware of the problems that you and the community are facing, and are sincerely apologizing.

Due to the merger of Infineon Technologies and Cypress Semiconductors, this issue already drew more attention within our company and increased our awareness of the growing importance of community contributions.

We are about to enhance our setup and have allocated dedicated resources to ensure regular maintenance and evaluation of contributions from you and other members of the community. 
The contributions that you have provided are very valuable. We already are evaluating them and prepare for a release of a new library.

Regrettably, that still will take several weeks, but we can assure you that we will continue the support for this repository.

Best regards,
Julian

jaenrig-ifx, 2021-09-20:
Done
https://github.com/Infineon/XMC-for-Arduino/releases/tag/V1.6.0

closed 2021-09-20 by jaenrig-ifx

techpaul, 2021-09-20:
@jaenrig-ifx Thanks for the overdue release, understand the upper management issues delaying things.