Infineon/XMC-for-Arduino #179: fetched archive size differs from size specified in index on mac

orottler, 2021-12-25:
2.0.0.-rc2 on mac Error: 13 INTERNAL: Cannot install tool Infineon:arm-none-eabi-gcc@5.4-2016q3: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index

works for ubuntu, but seems not connectíng to the USB port /dev/ACM03

techpaul, 2021-12-27:
I think you are in the worng place "2.0.0.-rc2" is a version not here as V1.60 is latest here for XMC-for-Arduino

If using the public XMC-for-Arduino get release V1.6.0 to try

I suspect you are using something different that happems to use the same compiler

Either that or you have access to something that is not on this Repo

TheBlackfurGuy, 2022-05-11:
I have this issue aswell, `2.0.0.-rc2` refers to the version of the Arduino IDE. Arduino started to develop a new version of their IDE based on VisualStudioCode. It has general feature parity with the "old" Arduino IDE but this issue also plaques me on my mac. I suspect the compiler artifact is not built for macos.

TheBlackfurGuy, 2022-05-11:
This also happens with the old (1.8.*) IDE

closed 2023-04-06 by jaenrig-ifx

jscottsf, 2024-03-05:
Having the same issue. Can't install the boards. Can this be reopened and resolved?

```
Downloading packages
Infineon:arm-none-eabi-gcc@5.4-2016q3
Infineon:XMCFlasher@1.2.1
Infineon:arm@1.4.0
Installing Infineon:arm-none-eabi-gcc@5.4-2016q3
Failed to install platform: 'Infineon:arm:1.4.0'.
Error: 13 INTERNAL: Cannot install tool Infineon:arm-none-eabi-gcc@5.4-2016q3: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 96949257 != 96956416
```