Infineon/XMC-for-Arduino #181: Arduino IDE Is XMC4800 Board supported ?

ValentinBousquie, 2022-02-28:
Hello,

Recently, I bought 2 Infineon XMC4800 Relax Kit to test a little EtherCAT network. To start the development, I tried to use XMC for Arduino Library with Arduino IDE. When I select the board on the IDE, i don't find the XMC4800 kit. However when i look into the variants folder I can see all the files required (XMC4800.h, pins_arduino.h ...). 

If i try to use EtherCAT by selecting the XMC4700 Relax Kit, ECAT0 is not defined so I can't use the xmc_ecat.h functions.

Just have the following question : Is this XMC4800 Relax Kit supported by Arduino IDE or not yet ? 

Thanks in advance for your help.

jaenrig-ifx, 2022-03-01:
Hi @ValentinBousquie,

It is not yet supported, and currently not planned.

ValentinBousquie, 2022-03-01:
Hi @jaenrig-ifx 

Thank you for your answer, and so is there any workaround in which i can use the XMC4800.h file instead of the XMC4700.h file ?

ValentinBousquie, 2022-03-01:
Yes, I tried this and I got compilation errors on the file **pins_arduino.h** so you were right.
Thanks again for your help.

jaenrig-ifx, 2022-03-02:
Yes, the right instantiation for the XMC4800 Relax Kit peripherals (gpios, adc, etc.) has to be done in such file. It seems that was just copied from the XMC4700 but not completed.

ValentinBousquie, 2022-03-03:
Yes, I saw it. It's not ready yet. I just tried to copy some missing parts from the XMC4700 pins_arduino. And it has compiled. However I can't trust the global functioning of this way.

closed 2022-03-03 by ValentinBousquie