Infineon/XMC-for-Arduino #182: Access to the GPIOs of connectors X1 and X2 using PlatformIO

JoseOlin, 2022-05-20:
Hi,

We are using an Infineon **XMC4700** board with **PlatformIO 6.0.1** 

We would like to know if there are plans to implement **access to GPIOs of the X1 and X2 pin headers**, since the [wiki](https://github.com/Infineon/XMC-for-Arduino/wiki/XMC4700-Relax-Kit)  says that only the Arduino shield pins are implemented.

Is it currently possible to use the XMCLib functions from PlatformIO?

Thanks in advance!

techpaul, 2022-05-20:
Speaking as a CONTRIBUTOR here

The PlatformIO limitation is down to PlatformIO and is most likely mentioned in the PlatformIO Wiki and may be limited in their implementation. You will have to probably use arduino pin methods to acces the pins.

Yes you can, but I personally do not know how well tested this has been. There is definition in the XMC-for-Arduino configuration file (pins_arduino.h for ***92*** arduino style pins including 22 ADC channels.

To do this you need to create a amatrix which I have been meaning to do as documentation for myself for some time. This requires these initial steps so you have a map of the Arduino pin to X1 and X2 pin numbers

1.  Go to the XMC-for Arduino Wiki page (https://github.com/Infineon/XMC-for-Arduino/wiki/XMC4700-Relax-Kit) and got down to  the section marked "PCB Design Data" and get the board design files.
2. In thezip file when unpacked there is a schematic file as PDF as well as Eagle

    Schematic_XMC4700_XMC4800_Relax_Kit_Series-V1.pdf

3. Page 3 of the schematic at the top shows the connectors for X1, X2 and Uno style shield 

4. On your computer, find the directory 'Arduino15 a nd locate BELOW that
"packages\Infineon\hardware\arm\1.6.0\variants\XMC4700\config\XMC4700_Relax_Kit"

  or from this repo

https://github.com/Infineon/XMC-for-Arduino/tree/master/variants/XMC4700/config/XMC4700_Relax_Kit

5. Open in text editor or IDE "pins_arduino .h" around line 156 you will see the code block STARTING

~~~
const XMC_PORT_PIN_t mapping_port_pin[] =
{
    /* 0  */    {XMC_GPIO_PORT2, 15},  // RX                           P2.15
    /* 1  */    {XMC_GPIO_PORT2, 14},  // TX                           P2.14
    /* 2  */    {XMC_GPIO_PORT1, 0},   // GPIO / External INT 0        P1.0
    /* 3  */    {XMC_GPIO_PORT1, 1},   // PWM40-2 / External INT 1     P1.1
    /* 4  */    {XMC_GPIO_PORT1, 8},   // GPIO                         P1.8

....
	//Additional pins for port X1 starting here
	/* 36  */   {XMC_GPIO_PORT3, 4},   // PWM42-2                      P3.4

......
	//Additional pins for port X2 starting here
	/* 70  */   {XMC_GPIO_PORT3, 13}, // SPI_2 SCLK                    P3.13

~~~
First ccolumn of commentd number gives you the arduino port number
Llast column of text Px.y gives the number of the port on the processor and often signal name on the schematic

Mapping the pin numbers and signal names  on the X1 and X2 connectors with this data will give you the pinout of the X1 and X2 connectors.

I would be interested in your results as this table is something I have been wanting to do for some time but not had the spare time to do.

techpaul, 2022-05-20:
Quick Investigation reveals
***
1. ALL X1 and X2 signals are at 3V3 levels
2. Do NOT connect 5V signals to these pins
3. Analog inputs are 3V3 maximum ***

Only the DIGITAL signals on the Shield connector are converted to 5V levels

Quick CSV of the signals to pins attached ***USE WITH CAUTION AT YOUR OWN PERIL***

techpaul, 2022-05-22:
I have updated my wiki to cover this with various notes

[XMC4700 Relax Kit X1 and X2](https://github.com/techpaul/XMC-for-Arduino/wiki/XMC4700-Relax-Kit-X1-and-X2 "XMC4700 Relax Kit - Additional I/O connectors")
Including [CSV file](https://github.com/techpaul/XMC-for-Arduino/blob/master/resources/wiki/xmc4700-X1-X2.csv "XMC4700 Relax Kit - X1 and X2 pinout") which might help
Other useful page
[XMC4xxx Series Analog pins](https://github.com/techpaul/XMC-for-Arduino/wiki/XMC4xxx-Series-Analog-pins "XMC4xxx series analogue pin WARNINGS")

This is somewhat delayed as I need the XMC4700 details and tried to do XMC4400 Platform2Go as well but that is a mess and has many errors and mistakes in it (wiki page for that).

Look out for a pull request to develop to update various pin_arduino.h for variants to tidy them up and make it easier to maintain as well as better comments

JoseOlin, 2022-05-22:
Thanks a lot @techpaul ! You have given us a lot of light!

I confirm that we can access pin 1 of the X2 header with the following code.

```
#define X2_1 70 // P3_13

void setup()
{
    pinMode(BUTTON1, INPUT);

    pinMode(LED1, OUTPUT);

    pinMode(X2_1, OUTPUT);
}

void loop()
{
    int button1Val = digitalRead(BUTTON1);
    digitalWrite(LED1, button1Val);

    digitalWrite(X2_1, button1Val);

}
```

Let me know if we can help with anything for the matrix you mention, as I can't find where the compiler binds 70 to `{XMC_GPIO_PORT3 , 13}`, other than line 233 where 70 is just a comment.

At the moment we will solve it with a header file with the defines in XN_P format
N: 1 or 2
P: Pin number from 1 to 40.

As soon as we verify that they all work correctly we share it in case someone else needs it while the documentation is updated.

Thanks again for your help!

techpaul, 2022-05-22:
The code is in "Arduino15/arm/v1.6.0/variants/xmc4700/config/XMC4700_Relax_Kit/pins_arduino.h" on local computer

[Master pins_arduino.h on repo](https://github.com/Infineon/XMC-for-Arduino/blob/master/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h)

The code has NO knowledge of connectors (that is hardware and could be on many connectors), but knows about Port pins in CPU and Arduino pins so
~~~
/* 70  */ {XMC_GPIO_PORT3, 13}, // SPI_2 SCLK                   P3.13
~~~
Binds Arduino pin 70 to the port pin. The 70 in a comment is the INDEX into the array, as it is the 71st line in a 0 index array that starts at line 137
~~~
const XMC_PORT_PIN_t mapping_port_pin[] =
    {
    /* 0  */    {XMC_GPIO_PORT2, 15},  // RX                           P2.15
....
~~~
Which gives the Arduino Pin number by indexing. It is a way of reducing the data storage used as the extra item in each row takes up space in Flash and Speeds up reading the array, by doing it that way
.
The "comment" is a way of helping developers to know which line of the array they are on.

techpaul, 2022-05-22:
I might have to use X1 and X2 connector on upcoming projecst as I need LOTS of GPIO and probably 16 ADC, several serial ports  and may need to review how many interrupts the board supports.

Confirmation of results would be appreciated even if  not all pins, I may release an additional update with X1 and X2 pins after port number in pins_arduino.h

JoseOlin, 2022-05-22:
> The 70 in a comment is the INDEX into the array, as it is the 71st line in a 0 index array that starts at line 137

Oh, I see! Thats clever.

I like that the elegance of efficiency isn't lost in the embedded world.

> Confirmation of results would be appreciated even if not all pins, I may release an additional update with X1 and X2 pins after port number in pins_arduino.h

We're going to use most of the odd pins from header X1 and the even ones from header X2, so we'll confirm at least these during the week.

techpaul, 2022-05-23:
For extra help I have added some more documentation to pins_arduino.h in MY repository fork see

https://github.com/techpaul/XMC-for-Arduino/blob/master/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h

Each line of X1 and X2 Arduino pins have the X1 or X2 pin number at end of line, for example
~~~
 /* 36  */ {XMC_GPIO_PORT3, 4}, // PWM42-2 / PWM12              P3.4     X1-1
~~~

JoseOlin, 2022-05-23:
Thanks @techpaul !

That definitely makes it clearer.

closed 2022-05-26 by techpaul