Infineon/XMC-for-Arduino #186: HardwareSerial conflicting declaration in pins_arduino.h

JoseOlin, 2022-05-24:
**Description**:

We are having problems enabling serial communication with a XMC4700 Relax Kit

The next minimal program:
```
#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
}

void loop()
{
    Serial.println("Hello world");
    delay(500);
}

```

Throws the errors:
```
/home/user/.platformio/packages/framework-arduinoxmc/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h:467: error: 'HardwareSerial' does not name a type; did you mean 'HardwareSerial_h'?
In file included from /home/user/.platformio/packages/framework-arduinoxmc/cores/Arduino.h:261:0,
                 from /home/user/.platformio/packages/framework-arduinoxmc/cores/HardwareSerial.h:36,
                 from /home/user/.platformio/packages/framework-arduinoxmc/cores/HardwareSerial.cpp:22:
/home/user/.platformio/packages/framework-arduinoxmc/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h:467:8: error: 'HardwareSerial' does not name a type; did you mean 'HardwareSerial_h'?
 extern HardwareSerial Serial;
        ^~~~~~~~~~~~~~
        HardwareSerial_h
```
and 

```
/home/user/.platformio/packages/framework-arduinoxmc/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h:432: error: conflicting declaration of 'void serialEvent()' with 'C' linkage
In file included from /home/user/.platformio/packages/framework-arduinoxmc/cores/Arduino.h:261:0,
                 from /home/user/.platformio/packages/framework-arduinoxmc/cores/Main.cpp:27:
/home/user/.platformio/packages/framework-arduinoxmc/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h:432:6: error: conflicting declaration of 'void serialEvent()' with 'C' linkage
 void serialEvent( ) __attribute__((weak));
      ^~~~~~~~~~~
```

**Target:**

Infineon XMC4700 Relax Kit

Linux Mint 20.2

PlatformIO Core, version 6.0.1

Thanks in advance!

JoseOlin, 2022-05-24:
@techpaul It only happens with the file from your fork, I was tinkering with it. May be I need some other files from your fork to make it work properly.

If I go back to the original pins_arduino.h it recognizes Serial but not Serial1.

How should I initialize Serial1 to access the second UART of the XMC4700?

JoseOlin, 2022-05-24:
Oh, now I see.

I was assuming that your implementation was [the one used in PlatformIO](https://github.com/Infineon/platformio-infineonxmc), but I see that they are two different projects.

Any recommendations to use your fork on PlatformIO?

techpaul, 2022-05-25:
I have no links with PlatformIO, they use some of the lower level for Arduino like operation. I have NO knowlledge of the workings of PlatformIO. This is a separate entity that uses SOME of the examples from here.

I note your link is to the Development version of Platform IO not the STABLE version.

For any further details and what PlatformIO does or limits things you will have to ask in PlatformIO as that is a separate entity

techpaul, 2022-05-25:
I had a quick scan of the PlatformIO and noted some of the examples are mising and some were DROPPED a few years ago as not being suitable or orther reasons.

I do wonder how up to date the inclusion of version changes in XMC-for-Arduino they have or have not done.

JoseOlin, 2022-05-26:
> I note your link is to the Development version of Platform IO not the STABLE version.

Yes, actually I tried both, but neither recognizes Serial1.

> I had a quick scan of the PlatformIO and noted some of the examples are mising and some were DROPPED a few years ago as not being suitable or other reasons.

I see. That explains why there is no example of a double UART.

> I do wonder how up to date the inclusion of version changes in XMC-for-Arduino they have or have not done.

The last release is almost a year ago, and the latest commits from two months ago. However the pins_arduino.h file is very similar to the one in this repository, but for some reason it doesn't recognize Serial1, even though
external HardwareSerial Serial;
external HardwareSerial Serial1;

appears in the file pins_arduino.h

I would use this implementation, however I find the Arduino editor very impractical for large projects.

Do you use an external editor or Arduino's own @techpaul ?

techpaul, 2022-05-26:
I generally do work with editing suites of files in Notepad++ and compile/flash serial monitor in Arduino IDE. I find Vsiual Studio, Eclipse and others bloaty, clunky and put even more "project" hierarchies and other quirks hampering. Few IDEs allow putting things in sensisble hierarchies for me and assume the person is ONLY writing code.

The multiple serial example is from over 2 years ago. The ability for two serials has been working longer than that.

JoseOlin, 2022-05-30:
The problem was caused by a manual change of a .h file. I think we can close this issue.

closed 2022-05-30 by JoseOlin