Infineon/XMC-for-Arduino #188: XMC-for-Arduino and PlatformIO for my own XMC4500 Platform

ProgData, 2022-05-30:
How I could read here: [Support for the XMC4500 Relax](https://github.com/Infineon/XMC-for-Arduino/issues/1) 
-> the XMC4500 Relax Kit will never be supported by Infineon because of missing VCOM.

I'm working here with my own XMC4500 Platform using DAVE and a Segger J-Link Debugger.
Is there a way to make "XMC-for-Arduino" in combination with PlatformIO be available for my Platform as I want to enable a small number of selected people to write little programs for it?

techpaul, 2022-05-30:
Speaking as a CONTRIBUTOR

Whilst PlatformIO is a SEPARATE project and has not updated its XMC-for-Arduino parts for a few years see #182 and #186

When playing with Dave are you using the VCOM (serial port) connected to a terminal emulator on your host PC?

I **personally** would like the platform supported but doubt it will happen

JoseOlin, 2022-05-30:
I subscribe in case a collaboration with PlatformIO comes into the plans.

Cheers!

techpaul, 2022-06-01:
If you look at the other boards supported the first Serial port connection is by default via the Debug USB connection. So when you connect to the host system via USB the device adds a COM xx or ttyxx (depending on OS) to the system.

The VCOM is a Virtual COM port via the USB to the host., from the Debug micro. I have no idea about the code in that or how to add the USB stack to XMC4500 to support Serial on VCOM from the onboard  USB directly connected to the main XMC4500.

This will only happen if Infineon choose to do it. Whilst I, as a CONTRIBUTOR  have no way to changer how the hardware or debugger portion is done.

ProgData, 2022-06-01:
@techpaul - ah ok,... so no compatible VCOM like the other Kits have -> no solution to make it work. That's all information I needed to decide how to go on. Thank you very much for the support and also for the work here!

closed 2022-06-03 by ProgData