Infineon/XMC-for-Arduino #206: RS485 Modbus libarary throwing errors

BrixInMotion, 2022-09-28:
Hi,
I want to read some registers from a frequency inverter using a XMC4700 Relax Kit and the Arduino Modbus library. Unfortunately there are some warnings, notes and an error related to the ArduinoRS485 library. The same messages appear when compiling an example sketch of the RS485 libary.

> C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp: In member function 'virtual void RS485Class::begin(long unsigned int, uint16_t, int, int)':
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:67:34: warning: invalid conversion from 'uint16_t {aka short unsigned int}' to 'XMC_UART_MODE_t {aka XMC_UART_MODE}' [-fpermissive]
   _serial->begin(baudrate, config);
                                  ^
In file included from C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/Arduino.h:256:0,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.h:23,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:20:
C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/HardwareSerial.h:75:10: note:   initializing argument 2 of 'void HardwareSerial::begin(uint32_t, XMC_UART_MODE_t)'
     void begin( uint32_t speed, XMC_UART_MODE_t config);
          ^
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp: In member function 'void RS485Class::sendBreak(unsigned int)':
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:163:29: warning: invalid conversion from 'uint16_t {aka short unsigned int}' to 'XMC_UART_MODE_t {aka XMC_UART_MODE}' [-fpermissive]
   _serial->begin(_baudrate, _config);
                             ^
In file included from C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/Arduino.h:256:0,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.h:23,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:20:
C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/HardwareSerial.h:75:10: note:   initializing argument 2 of 'void HardwareSerial::begin(uint32_t, XMC_UART_MODE_t)'
     void begin( uint32_t speed, XMC_UART_MODE_t config);
          ^
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp: In member function 'void RS485Class::sendBreakMicroseconds(unsigned int)':
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:173:29: warning: invalid conversion from 'uint16_t {aka short unsigned int}' to 'XMC_UART_MODE_t {aka XMC_UART_MODE}' [-fpermissive]
   _serial->begin(_baudrate, _config);
                             ^
In file included from C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/Arduino.h:256:0,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.h:23,
                 from C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:20:
C:\Users\me\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.6.0\cores/HardwareSerial.h:75:10: note:   initializing argument 2 of 'void HardwareSerial::begin(uint32_t, XMC_UART_MODE_t)'
     void begin( uint32_t speed, XMC_UART_MODE_t config);
          ^
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp: At global scope:
C:\Users\me\Documents\Arduino\libraries\ArduinoRS485\src\RS485.cpp:189:18: error: 'SERIAL_PORT_HARDWARE' was not declared in this scope
 RS485Class RS485(SERIAL_PORT_HARDWARE, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);

A. XMC 4700 Relx Kit 5V
B. Lib Version 1.6.0
C. Arduino IDE Version 1.8.13
D. ArduinoRS485 Version 1.0.2: https://github.com/arduino-libraries/ArduinoRS485
    ArduinoModbus Version 1.0.6: https://github.com/arduino-libraries/ArduinoModbus
    Hardware: MKR 485 shield (connected only TX and RX pins)

Can you tell me where I would need to define the SERIAL_PORT_HARDWARE ?
Thanks.

jaenrig-ifx, 2022-09-28:
Hi Kilian,

I believe those are not be defined in the XMC-for-Arduino, but you can use the Serial1 instance:
https://github.com/Infineon/XMC-for-Arduino/blob/43793053a5f1d86efe4fbf683aa835d251e05081/variants/XMC4700/config/XMC4700_Relax_Kit/pins_arduino.h#L476

You can add that that SERIAL_PORT_DEFINE to Serial1 and the rest of the constructor arguments (RS485Class RS485(SERIAL_PORT_HARDWARE, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);) in another #elif/#endif block here for the XMC4700 relax kit board: 
https://github.com/arduino-libraries/ArduinoRS485/blob/9a6b830880a578292388f1bd36eb2ec836d29f93/src/RS485.h#L31-L40

BrixInMotion, 2022-09-28:
Thanks Juan and thanks @techpaul ,

The #elif method worked quite nice to eliminate the errors in the RS485 library. 

> #ifdef __AVR__
#define RS485_DEFAULT_DE_PIN 2
#define RS485_DEFAULT_RE_PIN -1
#elif defined(XMC4700_Relax_Kit)
#define RS485_DEFAULT_DE_PIN 6
#define RS485_DEFAULT_RE_PIN 8
#define SERIAL_PORT_HARDWARE Serial1
#else
#define RS485_DEFAULT_DE_PIN A6
#define RS485_DEFAULT_RE_PIN A5
#endif

Just selected pins 6 and 8 as they are currently not occupied in my application.
However about 16 new errors came up originating from the Modbus library afterwards. I was able to fix them by commenting the TRUE / FALSE definition part in modbus.h:
https://github.com/arduino-libraries/ArduinoModbus/blob/ba5d0beaec03243d26258f4edca4748ef1c3a428/src/libmodbus/modbus.h#L45-L51

Now I can compile the complete ModbusRTUClientToggle.ino example sketch without errors but with the warnings left. Need to check, if this could cause trouble, but looks good so far :)

techpaul, 2022-09-28:
If you are running out of pins and your devices can output 3V3 levels or handle 3V3 inputs you can have lots more pins see

https://github.com/techpaul/XMC-for-Arduino/wiki/XMC4700-Relax-Kit-X1-and-X2

jaenrig-ifx, 2022-09-29:
Which warnings? 

The ones above seem to be due to the usage of custom [XMC types](https://github.com/Infineon/XMC-for-Arduino/blob/43793053a5f1d86efe4fbf683aa835d251e05081/cores/HardwareSerial.h#L36) in [begin()](https://github.com/Infineon/XMC-for-Arduino/blob/43793053a5f1d86efe4fbf683aa835d251e05081/cores/HardwareSerial.h#L75) vs Arduino AVR core [begin()](https://github.com/arduino/ArduinoCore-avr/blob/24e6edd475c287cdafee0a4db2eb98927ce3cf58/cores/arduino/HardwareSerial.h#L122).

BrixInMotion, 2022-09-29:
Only the warnings shown in my first post, yes.
Ahh okay then it should work. My only fear was, that a parameter declaration might exceed its memory allocation. Thanks :)

closed 2022-09-29 by BrixInMotion