Infineon/XMC-for-Arduino #222: Poor voltage amplitude on RX pin

BrixInMotion, 2022-12-06:
Hi,
I'm facing input issues with the RX pin (D0). I want to read the serial data sent from the RS485 shield. However, the RX pin on the XMC4700 5V Kit is always around 4,6V level even when nothing is connected to it. When D0 is connected with RX of the MKR shield, the line cannot reach 0V but only changes within 2,2V and 4,0V and thus it does not go below the "logical-off-threshold" of the XMC. 

A. XMC 4700 Relx Kit 5V
B. Lib Version 1.7.0 (also tested with 2.1.0.)
C. Arduino IDE Version 1.8.13
D. ArduinoRS485 Version 1.0.2: https://github.com/arduino-libraries/ArduinoRS485
Hardware: MKR 485 shield

<table>
    <tr>
        <td>[image]</td>
        <td>[image]</td>
    </tr>
    <tr>
        <td style="text-align: center">RX is disconnected from XMC board. This is how it should look like, with RX level between 0V and 3,5V.</td>
        <td style="text-align: center">RX is connected to XMC board. RX level is poor between 2,2V and 4,0V.</td>
    </tr>
</table>

For me it looks like there is a strong pull-up active inside the XMC / level shifter. Is there a another way to disable pullups by software? Did anybody face this issue before?

Thanks in advance!

BrixInMotion, 2022-12-06:
Edit: The use of `pinMode(0, INPUT)` or `pinMode(0, INPUT_PULLDOWN)` does not bring any changes.

techpaul, 2022-12-06:
Please also see closed issue #206 

First note the %v pins on XMC4700 Relax are through level converters with internal pullups. Changing the input type by pinmode() has no effect as that is determined by the Serial port configuration defined in pins_arduino.h, then that only affects the pin as far as the level converter.

See also
https://github.com/techpaul/XMC-for-Arduino/wiki/XMC4700-Relax-Kit-X1-and-X2
https://github.com/techpaul/XMC-for-Arduino/wiki/Analog-Functions-and-Additions
https://github.com/techpaul/XMC-for-Arduino/wiki/I2C-Analog-pins

Second Note the MKR 485 shield is meant to be plugged into an Arduino MKR CPU board

Thrid Note - Some of the pins used on the MKR board  such as A0 and A5 are **NOT** the same on XMC4700 as the MKR board is 5V on ALL pins wheres as A0 and A5 on an XMC4700 are 3V3. I suspect you are driving a 3V3 signal into a 5V device causing non-deterministic effects. Also note A5 on most boards has a connection to I2C lines nd has detremental effects see

https://github.com/techpaul/XMC-for-Arduino/wiki/XMC4xxx-Series-Analog-pins
https://github.com/techpaul/XMC-for-Arduino/wiki/I2C-Analog-pins

and XMC4700 relax kit wiki page

https://github.com/Infineon/XMC-for-Arduino/wiki/XMC4700-Relax-Kit

To use a MKR 485 board and the ArduinoRS385 library needs careful wiring and defining of pin functions to the FEW 5V pins you have on a XMC4700 5V relax kit.

Also note you CANNOT use the Buck converter to provide power onto X2 pins as the VIN is nowhere near the expected 7 to 12V on most Arduino boards

closed 2022-12-07 by BrixInMotion

techpaul, 2022-12-07:
That and the TWO 10k2 pull downs means it is a potential divider to produce 2/3 output max, but with other issues is not good.

I would expected an isolating pass through termination of 47R to 100R .

Age old case of works on their specifc setups only.