Infineon/XMC-for-Arduino #224: XMC Flasher not working

JensKapitza, 2023-01-07:
Problem:  64Bit Arduino will not flash XMC2GO

Fix: delete all 32Bit libs and link them to 64bit.

where can i change the search path? on arduino?

System is 64Bit with latest arduino on JAVA 11 64Bit

/opt/SEGGER/JLink_V784a$ ls -la *_x86*

libjlinkarm_x86.so -> libjlinkarm_x86.so.7

libjlinkarm_x86.so.7 -> libjlinkarm_x86.so.7.84.1

libjlinkarm_x86.so.7.84.1

kapitza@kapitza-desktop:/opt/SEGGER/JLink_V784a$ rm *_x86*

kapitza@kapitza-desktop:/opt/SEGGER/JLink_V784a$

ln -s libjlinkarm.so libjlinkarm_x86.so

kapitza@kapitza-desktop:/opt/SEGGER/JLink_V784a$

ln -s libjlinkarm.so libjlinkarm_x86.so.7

kapitza@kapitza-desktop:/opt/SEGGER/JLink_V784a$

ln -s libjlinkarm.so libjlinkarm_x86.so.7.84.1

techpaul, 2023-01-07:
Did you look at Wiki and previous issue

See wiki page https://github.com/Infineon/XMC-for-Arduino/wiki/Linux-64-bit-Flasher-to-board

Personally I have had Arduino IDE woeking on 64 bit Linux and many boards.

JensKapitza, 2023-04-05:
Problem is not the 64bit Linux it is the 64bit Arduino which finds the wrong path

techpaul, 2023-04-06:
If you look at the link above it shows how to make the correct PATH

boramonideep, 2023-06-29:
Thanks @techpaul for the support. 

@JensKapitza  Please open a new ticket if the issue still persists.

closed 2023-06-29 by boramonideep