Infineon/XMC-for-Arduino #241: Can not connect xmc2go to arduino ide

timmornYE, 2023-06-04:
When I connect my xmc2go to usb both red leds and the green led flash like wild, but I can get no connection to Arduino IDE. I installed J-Link and added xmc boards through adding https://github.com/Infineon/XMC-for-Arduino/releases/latest/download/package_infineon_index.json

But there appears no com port in windows and also journalctl does not show anything in Linux when connecting. When I try to upload also J-Link does not find USB connection. Only LEDs keep flashing like wild. 

Any suggestions to get my xmc2go working?

jaenrig, 2023-06-07:
Hi @timmornYE ,

That LED wildness seems to be the factory demo application :) First you have to be make sure that JLink is installed and added to the system path. 
If that is the case, you should be able to see a SEGGER serial device in the device manager in Windows, or the /dev/ttyACM* in Linux. 

Then, it should be as well displayed in the Arduino IDE as well.

timmornYE, 2023-06-10:
Thanks for information. There was a problem with the USB cable. Changed it and all worked fine.

closed 2023-06-10 by timmornYE