Infineon/XMC-for-Arduino #243: Defining SERIAL_ONBOARD raises error

ststolz, 2023-06-11:
https://github.com/Infineon/XMC-for-Arduino/blob/674ee375decfb3751c99ef283bfbccd1cf8fd3ee/variants/XMC1100/config/XMC1100_XMC2GO/pins_arduino.h#L64

Got `error: #elif with no expression` on this line when setting `#define SERIAL_ONBOARD` in my code using platformio. Think it should be used like this? Because I found no documentation on using onboard serial.

Fixed it with `#elif defined SERIAL_ONBOARD`.

techpaul, 2023-06-11:
Looks like most of the board variants need that fix as few people have used serial onboard, ceratinly XMC1 series

Thank you spotting that and the fix

boramonideep, 2023-06-16:
Hello @ststolz , in order to use the ONBOARD Serial instance, you just need to select the appropriate option from the dropdown shown below. This will set/define all the necessary macros automatically. There are no additional steps necessary from the user's side. Please let us know if you need further assistance. 
[image]

ststolz, 2023-06-18:
Like described in #245 I am using platformio. There was no howto to do it there, but I provided some information after I managed to set it up.

boramonideep, 2023-06-19:
Sorry for overlooking that the issue was occurring for PlatformIO. I hope it worked out with the manual addition of the correct build flags.

closed 2023-06-19 by boramonideep