Infineon/XMC-for-Arduino #245: Update xmc2go Wiki with onboard serial information

ststolz, 2023-06-12:
**Is your feature request related to a problem? Please describe.**

It is not trivial to set up onboard serial on xmc2go.

**Describe the solution you'd like**

Please update [xmc2go Wiki](https://github.com/Infineon/XMC-for-Arduino/wiki/XMC-2Go) with information how to set up onboard serial.

To contribute, I prepared a [minimal example in a gist](https://gist.github.com/ststolz/b644cc12855bbce8a4d1a07e1a690ed2).

ststolz, 2023-06-12:
I understand. At least I would have been happy to get the information what build Flag I have to change. I think this information should be offered by Infineon.

boramonideep, 2023-06-16:
Thanks @techpaul 

As described in https://github.com/Infineon/XMC-for-Arduino/issues/243#issuecomment-1594510418 , only the proper option has to be selected in the dropdown in the `tools` menu. The rest of the macros and build flags are set automatically.  

We shall be adding documentation regarding this to make this a little clearer.

boramonideep, 2023-06-19:
Brief info added here: https://github.com/Infineon/XMC-for-Arduino/wiki/Serial-selection

closed 2023-06-19 by boramonideep