Infineon/XMC-for-Arduino #258: Flashing failed: Failed to initialize BridJ error on MacOS

sskelapure, 2023-10-25:
Using XMC2Go 1100 board with JLink already installed;
JLink version - v7.92k
[image]

juuulianx, 2023-11-16:
I had the same issue on my M1 Pro MacBook. It worked then I used PlatformIO in VS Code, havent had any issues by now.

jscottsf, 2024-03-05:
Same issue on an M1. Can't really use the board now. Maybe I'll explore PlatformIO if there's a workaround. Fingers crossed.

```
Sketch uses 3416 bytes (5%) of program storage space. Maximum is 65536 bytes.
Global variables use 1784 bytes of dynamic memory.
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Mac OS X
Operating System: Mac OS X

Initialisation
--------------------------
Mac OS X
Loading JLink Library... 
java.lang.RuntimeException: Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: /private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib: dlopen(/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib, 0x0001): tried: '/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (no such file), '/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')))
	at org.bridj.Platform.initLibrary(Platform.java:444)
	at org.bridj.Platform.<clinit>(Platform.java:228)
	at org.bridj.BridJ.register(BridJ.java:191)
	at jlinkarm.JLinkARMLibrary.<clinit>(JLinkARMLibrary.java:26)
	at Infineon.JLinkDebugger.GetDLLVersion(Unknown Source)
	at Infineon.XMCDevice.Program(Unknown Source)
	at Infineon.upload.main(Unknown Source)
Caused by: java.lang.UnsatisfiedLinkError: /private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib: dlopen(/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib, 0x0001): tried: '/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (no such file), '/private/var/folders/cf/2z5l1jx95y5f78c9tfb7mwy80000gn/T/BridJExtractedLibraries5637530366829056195/libbridj.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(Unknown Source)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(Unknown Source)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(Unknown Source)
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.load0(Unknown Source)
	at java.base/java.lang.System.load(Unknown Source)
	at org.bridj.Platform.initLibrary(Platform.java:420)
	... 6 more
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'int org.bridj.Platform.sizeOf_ptrdiff_t()'
	at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
	at org.bridj.Platform.<clinit>(Platform.java:232)
	at org.bridj.BridJ.register(BridJ.java:191)
	at jlinkarm.JLinkARMLibrary.<clinit>(JLinkARMLibrary.java:26)
	at Infineon.JLinkDebugger.GetDLLVersion(Unknown Source)
	at Infineon.XMCDevice.Program(Unknown Source)
	at Infineon.upload.main(Unknown Source)
Failed uploading: uploading error: exit status 1
```