Infineon/XMC-for-Arduino #266: incorrect way to open application with python

ninjaboy667, 2024-01-30:
**Description**:
PC is running on local admin.

Code was pulled straight from the XMC 1100 sleep example file:

https://github.com/Infineon/XMC-for-Arduino/tree/master/libraries/DeviceControlXMC/examples/SleepModeXMC1100

[image]

**Steps to Reproduce:**

1. install Jlink, Arduino IDE, install XMC board manager 3.0.0 on arduino

2. Copy example files, attempt to upload

3.

**Expected Result:**

permission denied

**Actual Result:**

**Frequency (1/1)

Windows 10

Arduino IDE 2.2.1

ninjaboy667, 2024-01-30:
**Fixed**
Please add in the documentation that the Arduino IDE needs to Run as administrator in order to use the XMC-flasher.py.
A prompt could probably be added, or there surely is some way to run this without admin privileges ?

ederjc, 2024-01-31:
Hi @ninjaboy667,
Thanks for submitting & elaborating on this issue.

We will definitely look into this and improve the docs, or better - find a way to run this without admin privileges.

Best regards
Julian

per1234, 2024-08-06:
The problem is this code:

https://github.com/Infineon/XMC-for-Arduino/blob/bed41f488aa76eeb292c2e4117e0478e4d2c3e6b/tools/xmc-flasher.py#L71-L74

Is creating the `cmd.jlink` file in the current working directory. That is whatever path Arduino IDE was started from. It is common to install Arduino IDE in a path like `C:\Program Files\arduino-ide` and Windows imposes special security restrictions on this path, which causes the creation of the file to fail with this "`Permission denied`" error.

The solution is to adjust the `xmc-flasher.py` code so that it sets an absolute path for the `cmd.jlink` file instead of dumping it into whatever happens to be the current working directory. An appropriate path should be used for this purpose. Since this is a temporary file, it should be created under the system temporary folder.

LinjingZhang, 2024-08-12:
Hi @per1234 

Thank you very much for this hint.
Yes, the download path creates permission problems if the Arduino IDE is installed for all users of Windows. We will fix this in the next release. @ninjaboy667 

BR,
Linjing

LinjingZhang, 2024-09-10:
This was resolved in release 3.3.0, so close this issue.

LinjingZhang, 2024-09-10:
This was resolved in release 3.3.0, so close this issue

closed 2024-09-10 by LinjingZhang