Infineon/XMC-for-Arduino #267: Winreg required, not available on Linux

eChirp, 2024-01-30:
Since commit https://github.com/Infineon/XMC-for-Arduino/commit/abdeaa476b74298ee6b99bc89a51a6c997c3cd28 the `winreg` python module is required. This is not available on Linux.

```
Arduino: 1.8.19 (Linux), Board: "XMC1100 XMC2Go, PC, None"

Sketch uses 3324 bytes (5%) of program storage space. Maximum is 65536 bytes.
Global variables use 1780 bytes of dynamic memory.
Traceback (most recent call last):
  File "~/.arduino15/packages/Infineon/hardware/xmc/3.0.0/tools/xmc-flasher.py", line 1, in <module>
    import argparse, subprocess, os, sys, re, warnings, winreg
ModuleNotFoundError: No module named 'winreg'
```

Will Linux still be supported in the future?

ederjc, 2024-01-31:
Hi @eChirp,
Thanks for letting us know about this issue.

**Yes, Linux will still be supported in the future.** We will work on solving this issue, soon. In the meantime I would suggest to use [v2.2](https://github.com/Infineon/XMC-for-Arduino/releases/tag/V2.2.0).

Best regards
Julian

ederjc, 2024-02-21:
This issue has been fixed in PR #268. The fix will be part of the next release.

closed 2024-02-21 by ederjc